Ameba-AIoT/ameba-arduino-pro2 #32: AMB82-MINI compilation error.

Oliver0804, 2023-01-02:
[image]
Using the RTSP stream in the example...(StreamRTSPSingleVideoWithAudio)
macOS
Arduino 1.8.19
AmebaPro2 4.0.1-build20221222

Oliver0804, 2023-01-02:
'''
<built-in>: fatal error: opening dependency file /dev/null.d: Operation not permitted
ResolveLibrary(opening dependency file /dev/null.d)compilation terminated.

  -> candidates: []
exit status 1
'''

S10143806H, 2023-01-03:
Dear @Oliver0804
First of all, the stable version of Arduino SDK is on our `main` branch, kindly noted that the `dev` branch is still under development.
Secondly, if you encounter any similar error in the downloaded `main` branch Arduino SDK, help build an empty example .ino first due to cached compiled data will cause the compilation to fail.
Thank you.

M-ichae-l, 2023-01-06:
Closed. As there is no further issue.

closed 2023-01-06 by M-ichae-l