Ameba-AIoT/ameba-arduino-pro2 #74: Ethernet TCP Examples

ajsb85, 2023-03-29:
Dearest.

Could you implement the TCP examples using the Ethernet HAL API?

https://www.amebaiot.com/en/amebapro2-amb82-mini-arduino-server-client/

It is very important for our roadmap.

Thank you in advance.

Reference:

[ambpro2_sdk/component/mbed/targets/hal/rtl8735b/ethernet_api.c](https://github.com/ambiot/ambpro2_sdk/blob/main/component/mbed/targets/hal/rtl8735b/ethernet_api.c#L3)

S10143806H, 2023-03-30:
Dear @ajsb85 ,

Thank you for raising this issue and supporting Ameba Pro2 in your roadmap. 
We will get back to you again upon internal discussion.

Thanks.

S10143806H, 2023-03-30:
Hi @ajsb85 ,

So far there is no Ethernet hardware interface on the AMB82-MINI evaluation board. 
After talking about it, we won't be able to support the ethernet feature yet. 

Thanks

M-ichae-l, 2023-04-12:
Closed as AMB82 mini did not support the Ethernet. It is hardware limitation.

closed 2023-04-12 by M-ichae-l

TeaBear5, 2024-10-15:
> Closed as AMB82 mini did not support the Ethernet. It is hardware limitation.

We are using the AMB82 mini as a prototype device and would like to support ethernet to the RTL8735B chip when we move to our own board using the chip.  It appears that the underlying chip does in fact have support for the ethernet phy if my understanding is correct?

The picture below is taken from the datasheet.

[image]

Will we be able to support this setup as suggested below?

[image]

Is there firmware in this repo to support this already?  Additionally if there isn't, is it something that is possible?