Ameba-AIoT/ameba-arduino-pro2 #87: [Linux(Ubuntu 22.04)] SDK problems: 2. Ommiting "ommon_nn_models" because 'cp -r' is not specified

Sunyoup, 2023-04-16:
Hi
I bought AMB82-MINI boards, and I tried several example sketches from AMB Pro 2 SDK 4.0.2 in Arduino IDE.
All of them are working good, when I do so ON WINDOWS.
But when I do same ON LINUX(Ubuntu 22.04 64bit), I got this this message on Arduino IDE(1.8.19):

... cp: -r not specified; omitting directory '.../.arduino15/packages/realtek/hardware/AmebaPro2/4.0.2/variants/common_nn_models'

Please check this out.
Thanks in advance.

[image]

Sunyoup, 2023-04-17:
Hi @S10143806H 

Instead of showing the full compilation log, I present you the solution I found myself:

1. Open one of the C++ source code file:
~/.arduino15/packages/realtek/tools/ameba_pro2_tools/1.1.2/misc/src/postbuild_linux.cpp

2. Edit the line 179 as follows:
179 ...  cmdss << "cp " << common_nn_models_path << "* ./";
-> 
179 ... cmdss << "cp -r  " << common_nn_models_path << "/* ./";

3. Compile it:
 g++ postbuild_linux.cpp -o postbuild_linux

4. Replace  ~/.arduino15/packages/realtek/tools/ameba_pro2_tools/1.1.2/postbuild_linux executable.

After that, the NN stuffs are included correctly into the result and the Face Detection & Recognition, Object Detection examples are working good.

I prefer Linux because the project build job is much faster.
So I hope this be included in the next release.
And I think that the MacOS case should be looked into as well.

Sunyoup, 2023-04-18:
Hi @S10143806H 

I think it will be better for you to raise a PR. I got not enough extra time, busy for my own job.
Sorry but thank you.

S10143806H, 2023-04-19:
Hi @Sunyoup ,
The reason for asking you to PR is simply I don't want to take credit from you. Since this is an open-source repository, with your PR, you can become a contributor to this repository.

Anyway, before making the PR, I would like to confirm with you how to reproduce this issue?

When I tried to compile any of the NN examples on Ubuntu 22.02.4 with Arduino IDE 2.0 and running AmebaPro2 4.0.2 SDK. It seems like this compilation finished without showing your error/warning message. Please take a look at the screenshot below for more details: 

[image]

Thank you.

Sunyoup, 2023-04-19:
Hi @S10143806H 

1. I'm using Ubuntu 22.04 & Arduino IDE 1.8.19. Not 2.x.
    But I don't think it is the key issue.

2. Looking into your compilation log, I think that it is also a FAILURE case.
    It is saying that the sketch compile result is 8,658,944 bytes in size.
    The successful result should be greater than 14M bytes if the common_nn_* files are included correctly.
    My failure case was same at first, with making some sketch binary, flashing it, showing scenes via RTSP,
    but with NO FACE OR OBJECT DETECTION RECTANGLES.
    I don't know why your log does not show the "cp: -r not specified, omitting common_nn_ ..."
   (Maybe the command execution outputs are redirected >& /dev/null or something like that way.)

3. Actually I wanted & tried to raise a PR. But I failed.
    After a long work at my job place till late, I was very tired and
    before making a PR, cloning the repository onto my desktop failed due to the BIG size and
    temporary network problem. I will try it later again but you don't have to wait for it.
    (You told me that the development branch workers are already aware of it.)

4. Please don't forget the "/dev/null.d" problem as well.
5. The MacOS case can have same problems I think.

6. Anyway I think this AMB82-MINI board is AWESOME. I have tried esp32-cam, Freenove esp32s3-cam.
    But this is better than them I think. Thanks for developing & creating this masterpiece of product.

S10143806H, 2023-04-19:
Hi @Sunyoup ,

Yes, you are correct, this issue is IDE-independent, and my compiled image on have video features but not NN, causing the total image size is smaller.

Alternatively, without modifying the tool, you can just edit lines 142 & 144 in `platform.txt` to avoid this issue.

https://github.com/ambiot/ambpro2_arduino/blob/06b716d1da1c02bb4521e4383f75456fe42bdeb7/Arduino_package/hardware/platform.txt#L142

from: 
` recipe.objcopy.hex.pattern.linux="{ameba.tools_path}/{recipe.objcopy.hex.cmd}" "{ameba.tools_path}" "{recipe.objcopy.hex.imagepath_appntz}" "{compiler.path}" "{build.variant.path}/{build.sym_bklist}" "{recipe.objcopy.hex.imagepath_appntzmap}" "{build.upload_mode}" "{runtime.platform.path}/variants/common_nn_models" "{build.upload_mode_nn}" "{build.upload_mode_voe}" `
to
` recipe.objcopy.hex.pattern.linux="{ameba.tools_path}/{recipe.objcopy.hex.cmd}" "{ameba.tools_path}" "{recipe.objcopy.hex.imagepath_appntz}" "{compiler.path}" "{build.variant.path}/{build.sym_bklist}" "{recipe.objcopy.hex.imagepath_appntzmap}" "{build.upload_mode}" "{runtime.platform.path}/variants/common_nn_models/" "{build.upload_mode_nn}" "{build.upload_mode_voe}" `

(Add a '/' behind file path `{runtime.platform.path}/variants/common_nn_models`)

We will fix this issue in our `dev` branch by modifying the mentioned lines. 
You can also try from your side and get back to us.

----

Besides, the "/dev/null.d" problem have been fixed in `dev` branch:
https://github.com/ambiot/ambpro2_arduino/blob/8255ddc58469c85893b999d4bb68462a43527e27/Arduino_package/hardware/platform.txt#L123

Please do not worry about the PR, thank you for raising this issue and your support on AmebaPro2 :)

M-ichae-l, 2023-04-21:
This is resolved by  18469a1a9217cc0ff93382f3df913befe081f970

Sunyoup, 2023-04-23:
Closing this issue. I confirmed that this is resolved in 4.0.3

closed 2023-04-23 by Sunyoup