Ameba-AIoT/ameba-arduino-pro2 #89: 如何在AMB82-Mini 實現自定義UART(TX,RX) Serial1 or  Serial1 ports  TXRX 與arduino UNO Serial1 ports  TXRX 通訊交換資料?

3b012178, 2023-04-18:
最近買了amb82-mini 板子，但想要實現執行數個 Software serial ports，或者
想要同時併用 Hardware 與 Software serials，但是 Arduino 官方並不建議這樣作，
因為在一塊開發板上同時使用 Hardware+Software serial 或者執行多組的
Software serial，很容易產生傳輸瓶頸或亂碼，我試過後也是如此。

所以我想要先嘗試使用amb82-mini 板子 的 serial1 ports  TXRX， serial2ports  TXRX
但是目前官網上沒有寫如何使用到erial1 ports  TXRX， serial2ports  TXRX 只提供LOG_TX RX  ARDUINO 範例 但顯然沒辦法達到我的目的!

S10143806H, 2023-04-18:
您好 @3b012178，
感謝您使用amb82-mini開發板。目前我們正在准備新版本4.0.3的更新。
我們内部討論后是否將 UART 的支援加入開發 roadmap。

謝謝！

S10143806H, 2023-04-19:
@3b012178 ,
您可以嘗試以下程式碼測試三組 Hardware UART Serial pins (`PE1 PE2`, `PD15 PD16`, `PA2 PA3`) 的功能
```
/*
  Multiple Serial test

  Receives from the main serial port, sends to the others.
  Receives from serial port 1, sends to the main serial (Serial 0).

  This example works only with boards with more than one serial like Arduino Mega, Due, Zero etc.

  The circuit:
  - any serial device attached to Serial port 1
  - Serial Monitor open on Serial port 0

  created 30 Dec 2008
  modified 20 May 2012
  by Tom Igoe & Jed Roach
  modified 27 Nov 2015
  by Arturo Guadalupi

  This example code is in the public domain.
*/

void setup() {
  // initialize both serial ports:
  Serial.begin(115200);
  Serial1.begin(115200);
  Serial2.begin(115200);
  Serial3.begin(115200);
}

void loop() {
  // read from port 1, send to port 0:
  if (Serial1.available()) {
    int inByte = Serial1.read();
    Serial.write(inByte);
  }
  if (Serial2.available()) {
    int inByte = Serial2.read();
    Serial.write(inByte);
  }
  if (Serial3.available()) {
    int inByte = Serial3.read();
    Serial.write(inByte);
  }

  // read from port 0, send to port 1:
  if (Serial.available()) {
    int inByte = Serial.read();
    Serial1.write(inByte);
    Serial2.write(inByte);
    Serial3.write(inByte);
  }
}
```

M-ichae-l, 2023-04-21:
Closed as there is no further issue.

closed 2023-04-21 by M-ichae-l