Ameba-AIoT/ameba-arduino-pro2 #108: Please use library name same as library directory name

ouoam, 2023-07-02:
From code

https://github.com/arduino/arduino-cli/blob/fe6d7499eb5e0920419829371c0e11f9af000499/legacy/builder/unused_compiled_libraries_remover.go#L42-L46

Arduino builder will remove directory not in using library name.
So, Arduino builder will remove compiled library and rebuild every time.

By change library name saves me 5 second every compile.
When I using `Watchdog, RTC, Multimedia, WiFi, Wire, Http, FileSystem`.

M-ichae-l, 2023-07-10:
Hi @ouoam . The lib name is used to replace the default. For example, Ameba has its own WiFi lib which is  different from Arduino default. Therefore, to avoid conflicts and compile errors, same name WiFi will present to replace the default WiFi lib.

ouoam, 2023-07-22:
HI @M-ichae-l . I think Arduino not have default lib.
ESP32 [lib](https://github.com/espressif/arduino-esp32/tree/master/libraries) can have library name WiFi and Wire without any compile errors.
Arduino AVR [lib](https://github.com/arduino/ArduinoCore-avr/tree/master/libraries), STM32 [lib](https://github.com/stm32duino/Arduino_Core_STM32/tree/main/libraries), ESP8266 [lib](https://github.com/esp8266/Arduino/tree/master/libraries) can also have library name Wire and SPI.

If you want to use AmebaWire name, can you change directory name to AmebaWire

M-ichae-l, 2023-07-24:
https://github.com/arduino-libraries/WiFi

ouoam, 2023-07-24:
From https://arduino.github.io/arduino-cli/0.27/sketch-build-process/#dependency-resolution
Ameba library must win https://github.com/arduino-libraries/WiFi in step 4
> A library that is architecture optimized wins against a library that is not architecture optimized (see [Architecture Matching](https://arduino.github.io/arduino-cli/0.27/sketch-build-process/#architecture-matching))

Or if pass step 4 can win in step 5
> A library that has a better "location priority" wins (see [Location Priority](https://arduino.github.io/arduino-cli/0.27/sketch-build-process/#location-priority))

With sub step 4
> The library is bundled with the board platform/core ([{runtime.platform.path}/libraries](https://arduino.github.io/arduino-cli/0.27/platform-specification/#global-predefined-properties))
---
This is example when I change ameba WiFi library name and have https://github.com/arduino-libraries/WiFi library
[image]

M-ichae-l, 2023-07-24:
Hi @ouoam , 
The log above clearly shows that "Arduino" WiFi lib will replace by Ameba "WiFi" lib. 
Using same name will let IDE to processed to replace, that is Ameba SDK needs to do for avoiding conflicts.

closed 2023-08-15 by github-actions[bot]