Ameba-AIoT/ameba-arduino-pro2 #136: use i2c scanner issue..

Oliver0804, 2023-09-26:
``` // --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    https://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
// 
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

#include <Wire.h>

void setup()
{
  Wire.begin();

  Serial.begin(115200);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```
I used Arduino IDE 2.2 to run a simple i2c scanner code, but I didn't get the correct response. Can you check the result of this code execution?

Oliver0804, 2023-09-26:
I have also used the 'wire' from the Ameba2 mini examples, and verified with an oscilloscope that there is no output change in every GPIOs pin.

S10143806H, 2023-09-29:
Dear @Oliver0804 ,
Thanks for reporting this issue.
I have verified that AMB82-MINI does not support examples such as "I2C - Scan I2C devices" nor "I2C - Display Data on LCD_Screen" at this point of time, we will update you again upon deploying the bug fix.

Oliver0804, 2023-10-13:
keep

Oliver0804, 2023-10-30:
keep again...

Oliver0804, 2023-11-13:
Keep

S10143806H, 2023-11-21:
Hi @Oliver0804 ，

Example will be provided on v4.0.6 early release (dev branch) today, lower level including HAL and ROM code have been modified, thus we need to update the archive libraries as well. 

The glance of the code can be found below:
Open the example in “File” -> “Examples” -> “AmebaWire” -> “I2C_Scanner”.
 
[image]

When the I2C bus detect any I2C device, the serial monitor will show the address of the I2C device as shown below:
 
[image]

 
When there is no I2C device connected to the board, the Arduino IDE serial monitor will show the message below:
 
[image]

Thanks

Tomi-ren, 2023-12-05:
It has been tested and the problem still exists. When no i2c device is mounted, a pile of addresses will still be scanned

Tomi-ren, 2023-12-05:
I used amb82 for testing

Tomi-ren, 2023-12-05:
Thank you. After I updated the library functions at the same time, the problem was resolved

S10143806H, 2023-12-05:
@Tomi-ren,
Glad that your issue has been solved.

Tomi-ren, 2023-12-05:
> @Tomi-ren, Glad that your issue has been solved.

But I tried multiple i2C devices, but the amb82mini couldn't scan their addresses and kept saying "No I2C devices found". On the contrary, using Arduino Uno to run the scan example made it easy to scan the devices, which I find difficult to understand

Tomi-ren, 2023-12-06:
Reminder everyone, if using pin 12.13, please change Wire1. begin() to Wire. begin(), and you will find that it is working properly

S10143806H, 2023-12-06:
@Tomi-ren ,
Sorry that since it's the dev branch, while I was doing the commit I didn't change from Wire1 to Wire that might causing potential confusing. But anyway, glad that you have solved your problem, and I will proceed to close this issue.

closed 2023-12-06 by S10143806H

Oliver0804, 2023-12-29:
> Hi @Oliver0804 ，
> 
> Example will be provided on v4.0.6 early release (dev branch) [V4.0.6-QC-V01](https://github.com/ambiot/ambpro2_arduino/tree/V4.0.6-QC-V01), lower level including HAL and ROM code have been modified, thus we need to update the archive libraries as well.
> 
> The glance of the code can be found below: Open the example in “File” -> “Examples” -> “AmebaWire” -> “I2C_Scanner”.
> 
> ![image1](https://private-user-images.githubusercontent.com/9852678/284431226-b5999a91-ec37-4612-b7e5-d1e73066b7a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4MzAzMzUsIm5iZiI6MTcwMzgzMDAzNSwicGF0aCI6Ii85ODUyNjc4LzI4NDQzMTIyNi1iNTk5OWE5MS1lYzM3LTQ2MTItYjdlNS1kMWU3MzA2NmI3YTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMDYwNzE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTY4YTBkMzc2MTU4MzYwNDkzODExNTIxMjMzM2RkYjgwNmQyN2NiMWNmODA2MzY2ODA0MTNmYzUxMzhiYWJjOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.JmbjIXNSRMI_NFNgLM3-tJqBRX8gTWBkqmUzdCgYJPc)
> 
> When the I2C bus detect any I2C device, the serial monitor will show the address of the I2C device as shown below:
> 
> ![image2](https://private-user-images.githubusercontent.com/9852678/284431241-43c0d6e8-3185-4edf-8884-a058187b5242.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4MzAzMzUsIm5iZiI6MTcwMzgzMDAzNSwicGF0aCI6Ii85ODUyNjc4LzI4NDQzMTI0MS00M2MwZDZlOC0zMTg1LTRlZGYtODg4NC1hMDU4MTg3YjUyNDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMDYwNzE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzNmZDFjYWYzYTI4MTY4MjczNjM5NDhjZTNmOGZjNWY4ZjAxNTg2NWU3NGE5MDZiMjcwY2JkNGQzNjYyZTkzMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.i0DEQEmrHyQvOGq6H3yf2TY5PjH0uchUw3tbzECHihg)
> 
>   When there is no I2C device connected to the board, the Arduino IDE serial monitor will show the message below:
> 
> ![image3](https://private-user-images.githubusercontent.com/9852678/284431257-2dabecbf-141e-4372-8ff7-e08c2d583b6d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4MzAzMzUsIm5iZiI6MTcwMzgzMDAzNSwicGF0aCI6Ii85ODUyNjc4LzI4NDQzMTI1Ny0yZGFiZWNiZi0xNDFlLTQzNzItOGZmNy1lMDhjMmQ1ODNiNmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDIzMTIyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMjlUMDYwNzE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGRiMDBkN2ZkYzhhYjA0ZTU2ZWU1ZDlkMTYyMGM2NjVhNmI4NjllMTRiOGJmMzJkODZmOTllYjZhMmFlOGYxZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.tjvOyBMmUxYVahfpMSZUwEHI3NcByq-BTrAtW0PA_tQ)
> 
> Thanks

I will wait for it to be merged into the release version before I proceed with testing.

geofrancis, 2023-12-29:
> Hi @Oliver0804 ，
> 
> Example will be provided on v4.0.6 early release (dev branch) [V4.0.6-QC-V01](https://github.com/ambiot/ambpro2_arduino/tree/V4.0.6-QC-V01), lower level including HAL and ROM code have been modified, thus we need to update the archive libraries as well.

This is very good, i2c support has been a real issue with using this board for any projects.