Ameba-AIoT/ameba-arduino-pro2 #152: RTL8735BM programming not working with uartfwburn.linux, ping failure

dashxdr, 2023-12-01:
When I try to program an image to my device using

> ./uartfwburn.linux -p /dev/ttyUSB0 -f flash_ntz.nn.bin -b 115200 -U

I get errors like ping failure, as uartfwburn.linux is unable to confirm the device is present. I'm using just RX and TX without any hardware flow control. I know the device is in BootFromUART mode, here is the uart log:

> == Rtl8735b IoT Platform ==
> Chip VID: 0, Ver: 0
> ROM Version: v3.0
> 
> [Normal mode]
> BootFromNORFlash
> [SPIF Err]Invalid ID
> [SPIF Err]Invalid ID
> [BOOT Err]Flash init error (io_mod=0, pin_sel=0)
> Fallback to NAND Flash
> Boot from NAND Flash
> [BOOT Err]snafc init error
> Fallback NOR Flash
> BootFromNORFlash
> [SPIF Err]Invalid ID
> [SPIF Err]Invalid ID
> [BOOT Err]Flash init error (io_mod=0, pin_sel=0)
> Fallback to NAND Flash
> Boot from NAND Flash
> [BOOT Err]snafc init error
> Fallback NOR Flash
> BootFromNORFlash
> [SPIF Err]Invalid ID
> [SPIF Err]Invalid ID
> [BOOT Err]Flash init error (io_mod=0, pin_sel=0)
> Fallback to NAND Flash
> Boot from NAND Flash
> [BOOT Err]snafc init error
> Fallback NOR Flash
> BootFromUART
> Image Load over UART1 baud=115200

Since uartfwburn.linux is closes source I have no idea how to proceed. Is there any documentation on the protocol the RTL8735 uses during the uart boot process?

Thanks anyone for any help!
-Dave

dashxdr, 2023-12-04:
@M-ichae-l I was able to get the board into the usual flashrom programming mode and the uartfwburn.linux works when I use "-n pro2". Unfortunately the image isn't acceptable to the RTL8735,I get this error:

> == Rtl8735b IoT Platform ==                                                     
> Chip VID: 0, Ver: 0                                                             
> ROM Version: v3.0                                                               
>                                                                                 
> [Normal mode]                                                                   
> BootFromNORFlash                                                                
> [Start Boot ROM...]                                                             
> === Load CERTI ===                                                              
> [BOOT Err]ld_certi_idx err:-1                                                   
> [BOOT Err]StartUp@0x0: Invalid RAM Img Signature!Boot_err(0x11)                 

Is there any way to erase the certification requirement on the RTL8735? My guess is this is happening because of something written into its flashrom.

I just want to disable any CERT requirement if possible. Thanks!

M-ichae-l, 2023-12-05:
If you are using AMB82-mini, you can refer to https://www.amebaiot.com/en/amebapro2-amb82-mini-arduino-getting-started/ 
for "erase flash"
Other boards at this SDK can't.

dashxdr, 2023-12-05:
@M-ichae-l I think I can erase and program the 8 pin flash device, the trouble is the RTL8735 doesn't accept anything but the original image programmed to it. I need to be able to reprogram the flashrom of the RTL8735 itself to change its behaviour. I tried those options but the end result is the same... RTL8735 refuses to load my file.

M-ichae-l, 2023-12-05:
@dashxdr  
Do you mean that you have tried **Erase All Flash Memory (16MB)** under Arduino IDE with hardware AMB82-mini. And still cant upload the code.

dashxdr, 2023-12-05:
@M-ichae-l Yes I've tried that. The 8 pin 16 megabyte flashrom can be erased and rewritten. That's not the problem. The problem is the RTL8735 doesn't accept anything but the original image burned into the 16 megabyte flashrom.

This is a hardware device equivalent to the AMB82-MINI development board. I want to be able to reprogram it. Something has been done to this device that prevents me from loading my own code and having it executed.

Is there any way to prevent the RTL8735 boot process from checking the certificate?

M-ichae-l, 2023-12-05:
@dashxdr  Unfortunately the boot process can not be changed under this SDK. 
You may want to try the following suggestions. 
- There are some system data inside the flash. If your bin not build in the right way, the data may over write the system data. 
- Nor flash only for this image tool and SDK
- `This is a hardware device equivalent to the AMB82-MINI development board` doesn't sounds right to use this SDK.

closed 2023-12-19 by M-ichae-l