Ameba-AIoT/ameba-arduino-pro2 #154: FaceMask Detection

rkuo2000, 2023-12-03:
[Facemask dataset](https://www.kaggle.com/datasets/ashishjangra27/face-mask-12k-images-dataset) are 96x96 pictures, I have trained a CNN model file at [kaggle.com/code/rkuo2000/facemask-cnn](https://www.kaggle.com/code/rkuo2000/facemask-cnn/)

still need to run MTCNN or FaceDetection with camera's image, 
then crop the face and resize into 96x96 for detection.

Is it possible to modify these from your FaceDetection example ?
Or should I directly go for training YOLOv7 with facemask data ?

M-ichae-l, 2023-12-04:
Currently AMB82-mini only supports following models, 
[image]

M-ichae-l, 2024-02-13:
Yolov7-tiny .pt online transfer is supported now. [https://www.amebaiot.com/en/amebapro2-ai-convert-model/](https://www.amebaiot.com/en/amebapro2-ai-convert-model/)
Closed as issue resolved.

closed 2024-02-13 by M-ichae-l