Ameba-AIoT/ameba-arduino-pro2 #164: MPU6050 Library example compilation failed

rkuo2000, 2023-12-12:
I am using MPU6050 Library by Electric Cats

Its examples will have the following errors in compilation : dtostrf and printfloatx  are missing

WARNING: library MPU6050 claims to run on avr, samd, sam, esp8266, esp32, stm32, renesas, renesas_portenta, renesas_uno architecture(s) and may be incompatible with your current board which runs on AmebaPro2 architecture(s).
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp: In member function 'void MPU6050::PrintActiveOffsets()':
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3398:3: note: in expansion of macro 'printfloatx'
 3398 |   printfloatx("", Data[0], 5, 0, ",  ");
      |   ^~~~~~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3399:3: note: in expansion of macro 'printfloatx'
 3399 |   printfloatx("", Data[1], 5, 0, ",  ");
      |   ^~~~~~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3400:3: note: in expansion of macro 'printfloatx'
 3400 |   printfloatx("", Data[2], 5, 0, ",  ");
      |   ^~~~~~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3403:3: note: in expansion of macro 'printfloatx'
 3403 |   printfloatx("", Data[0], 5, 0, ",  ");
      |   ^~~~~~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3404:3: note: in expansion of macro 'printfloatx'
 3404 |   printfloatx("", Data[1], 5, 0, ",  ");
      |   ^~~~~~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3379:20: error: 'dtostrf' was not declared in this scope
 3379 |       Serial.print(dtostrf((float)Variable, Spaces, Precision, S)); \
      |                    ^~~~~~~
d:\Users\rkuo2\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3405:3: note: in expansion of macro 'printfloatx'
 3405 |   printfloatx("", Data[2], 5, 0, "\n");
      |   ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1

M-ichae-l, 2023-12-12:
`WARNING: library MPU6050 claims to run on avr, samd, sam, esp8266, esp32, stm32, renesas, renesas_portenta, renesas_uno architecture(s) and may be incompatible with your current board which runs on AmebaPro2 architecture(s).`

I believe this is the reason.

rkuo2000, 2023-12-12:
Look at the error messages,
It is missing avr/dtostrf, printfloatx
Please patch to fix it. Thanks!

closed 2023-12-12 by rkuo2000