Ameba-AIoT/ameba-arduino-pro2 #166: I2C Err when running MPU6050_DMP_Teapot.ino

rkuo2000, 2023-12-12:
sample code: 

[I2C  Err]terminate tx op. after error

[I2C  Err]residual len: 1

[I2C  Err]last dat: 68

[I2C  Err]tx abrt src:00000001

[I2C  Err]terminate tx op. after error

[I2C  Err]residual len: 2

[I2C  Err]last dat: 68

[I2C  Err]tx abrt src:00000001

[I2C  Err]terminate tx op. after error

[I2C  Err]residual len: 1

[I2C  Err]last dat: 68

[I2C  Err]tx abrt src:00000001

[I2C  Err]terminate tx op. after error

[I2C  Err]residual len: 1

[I2C  Err]last dat: 68

D
[I2C  Err]tx abrt src:00000001

[I2C  Err]terminate tx op. after error

[I2C  Err]residual len: 3

[I2C  Err]last dat: 68

MP Initialization failed (code 1)
Sample Rate: 100

M-ichae-l, 2023-12-13:
@rkuo2000 
This SDK does not support MPU6050 yet.

M-ichae-l, 2024-02-29:
@rkuo2000 
MPU6050 is supported by latest early release version 4.0.7

rkuo2000, 2024-03-03:
I confirmed MPU6050_raw & MPU6050_DMP6 were verified, MPU6050 is supported.

closed 2024-03-03 by rkuo2000