Ameba-AIoT/ameba-arduino-pro2 #168: MPU6050_raw.ino [ERROR] requestFrom: readed=1, quantity=14

rkuo2000, 2023-12-13:
Library: MPU6050 by Electronic Cats
MPU6050.h patch with #include <avr/dtostrf.h>

50,232,20713,4345,-7903,-4353
50,232,20713,4345,-7903,-4353

[ERROR] requestFrom: readed=1, quantity=14 

47,-16159,-14109,8193,4071,22749
47,-16159,-14109,8193,4071,22749

[ERROR] requestFrom: readed=1, quantity=14 

61,-18197,-18208,8440,20475,-4369
61,-18197,-18208,8440,20475,-4369

rkuo2000, 2023-12-13:
char *dtostrf (double val, signed char width, unsigned char prec, char *sout);

add the above code should fix compilation error of missing to include <avr/dtostrf.h>

M-ichae-l, 2024-02-29:
MPU6050 is supported by latest early release version 4.0.7

rkuo2000, 2024-03-01:
Yes, V4.0.7 build 20240329 has fixed this error. Thanks !

closed 2024-03-01 by rkuo2000