Ameba-AIoT/ameba-arduino-pro2 #170: Can I reduce the time it takes to initiate the camera?

jonathan84clark, 2023-12-13:
I'm working on an application that requires the device wakes up and takes a picture extremely quickly. So far I've been able to get this down to 267ms which meets spec for the project. But to do this I had to initialize the file system AFTER I initialized the camera. I need to be able to apply settings to the camera that are stored on the SD card. However reading the SD card takes an additional ~250ms which pushes the overall time to picture up to nearly 600ms which is a little longer than I want. I want to see if I can shave off a few more milliseconds.

One of the things I would like to look at is the camera boot process. Right now it shows this:
```
VOE]ext_in = 0 sync = 0
[VOE]md init success
[VOE]algo ver ea63d5a 
[VOE]pack_v 0x0000 0x0002 cus_v 0 iq_id 0 dn 0 day 0 night 1 other 2 offset 224 length 16436 iq_size 32716 
[VOE]Ver 0x0001000c Fast3A Cnt AE 1 AWB 1 period AE 5 AWB 1 delay 0 0 
[VOE][isp_mod_sensor_init][1581]Errhdr_mode 0 is over sensor driver num 1 
[VOE]hdr_mode 0 sensor driver num 1 
[VOE]fps: 1880018620, pclk: 0, hts: 1077805056
[VOE]fps max 30.000000 min 5.000000 
[VOE]exposure_step 29.629629
[VOE]change sensor mode => 1920x1080@30.000000fps - 'linear' 
[VOE]min_fps 5.000000, max_fps 30.000000, exp_step 29.629629
[VOE]md ver 0x6d640100 
[VOE]ae ver 0x61650200 
[VOE]awb ver 0x77620100 
[VOE]cur_hdr_mode = 0
[VOE]VOE MEM Size = 11447 Used=  3550 KB Free= 7897 KB (8086944)
[VOE]stream 0  buffer 0: 0x7054ac00 size 460800 
[VOE]stream 0  buffer 1: 0x705bb500 size 460800 
[VOE]first_config_osd2_block_num[0]: 1
[VOE]osd2_block_num[0]: 24
[VOE]NV12 640x480 1/30
[VOE]dynamic set fps 0 -> 30 ok
[VOE]sensor power on
[VOE]zoom crop default setting
[VOE]status == 1718
[VOE]buffer size == (0x7063bbc0) (115200 38400) queue(20)
```
I want to get into the code that generates this output to see what I can tighten up. However I can't seem to find the .c/.cpp files generating this output. I'm seeing header files (.h) but no matching .c or .cpp files. I am also seeing a lot of .a files which makes me suspect that some of the files are precompiled binaries and not source code. Are the source files used to compile those binaries available? Is there a way I could have access to them? Or am I just looking in the wrong place?

Thanks,

Jonathan L Clark

M-ichae-l, 2024-04-22:
@jonathan84clark 
For your situation, yes you are right, the boot process are prebuild as bin files for the SDK to use. For more sensors supported currently there are 2, GC5035 and JXF37. 
To go further, I think currently it is very hard to push the whole process faster. You may want to try running your application at multi-thread.

M-ichae-l, 2024-07-03:
there is a feature FCS, but have not port to Arduino SDK yet

M-ichae-l, 2024-10-07:
Closed as no further request.

closed 2024-10-07 by M-ichae-l