Ameba-AIoT/ameba-arduino-pro2 #173: YOLOv7 pothole detection

rkuo2000, 2023-12-14:
model file best.pt was trained by using [https://www.kaggle.com/code/rkuo2000/yolov7-pothole-detection](https://www.kaggle.com/code/rkuo2000/yolov7-pothole-detection)

[https://www.kaggle.com/code/rkuo2000/yolov7-pothole-detection](https://github.com/rkuo2000/EdgeAI-course/tree/main/pretrained/yolov7-pothole-detection)
PoltholeClassList.h
best.pt
yolov7_pothole-tiny.yaml

Problem : online conversion failed 2 times
best.pt renamed to best.weights
yolov7_pothole-tiny.yaml (config file)
a test.jpg was uploaded

M-ichae-l, 2024-01-23:
@rkuo2000 
The yolov7-tiny pt is going to be supported by [AMB82 AI Model Conversion
](https://www.amebaiot.com/en/amebapro2-ai-convert-model/)

rkuo2000, 2024-02-07:
YOLOv7-PyTorch will be supported in AmebaPro2 v4.0.7

M-ichae-l, 2024-02-13:
@rkuo2000 
not really, it is depends on the online transfer not SDK. It is already supported.

closed 2024-02-13 by M-ichae-l