Ameba-AIoT/ameba-arduino-pro2 #179: [Error] AmebaPro2 directory only allow 1 SDK!!! Please check again.

Oliver0804, 2023-12-21:
Problem:
Compilation failure
[image]

Environment description:
Arduino IDE version 1.8.19 https://raw.githubusercontent.com/ambiot/ambpro2_arduino/dev/Arduino_package/package_realtek_amebapro2_early_index.json

I have tried reinstalling, but some data might be stored in other cache locations in Windows.
Can you help provide information on which paths this program checks so that I can manually delete them?

pammyleong, 2023-12-22:
Hi @Oliver0804, please ensure that your AmebaPro2 folder only contains one SDK folder. C:\Users\{USERNAME}\AppData\Local\Arduino15\packages\realtek\hardware\AmebaPro2
[image]

Oliver0804, 2023-12-29:
[image]
Thank you. Indeed, there are two folders present. 
I'm not sure what might have happened, 
but it's clear that installing both Arduino 1.0 and 2.0 on my computer at the same time might have led to this situation

closed 2023-12-29 by Oliver0804