Ameba-AIoT/ameba-arduino-pro2 #180: Compatibility Issues and Error Messages When Installing Development Versions in Different Arduino IDEs

Oliver0804, 2023-12-21:
When installing using the same path https://github.com/ambiot/ambpro2_arduino/raw/dev/Arduino_package/package_realtek_amebapro2_early_index.json, the 4.0.6 development version can be seen in Arduino IDE version 1.0, but it's not visible in Arduino IDE 2.0. Also, when installing other development versions in Arduino 2.0, there are occasions where a 404 error message appears.

[image]
[image]

pammyleong, 2023-12-22:
Hi @Oliver0804,

I tried on Arduino IDE 2.2.0 and 2.0.0, and the path works fine for me. May I know which IDE version you are using now?
We have changed the link to remove ".com" as the latest IDE version doesn't allow ".com" in the path.

Maybe you can try the old link if you are using the older version:
Old: https://github.com/ambiot/ambpro2_arduino/raw/main/Arduino_package/package_realtek.com_amebapro2_index.json
New: https://github.com/ambiot/ambpro2_arduino/raw/main/Arduino_package/package_realtek_amebapro2_index.json

Else, you can also try deleting the .json file in Arduino15 folder and it will be installed again when you open Arduino IDE and input the path in Preference. 
[image]

M-ichae-l, 2024-01-23:
`404 server issue. 
`
Looks like anything wrong with your internet? or GitHub server issue.

M-ichae-l, 2024-02-13:
Close as there is no further issue.

closed 2024-02-13 by M-ichae-l