Ameba-AIoT/ameba-arduino-pro2 #181: I2C Wire.endTransmission resets I2C device

moschotto, 2023-12-23:
Hi,

when trying a super simple master/slave setup (AMB82-mini <> ESP32) i discovered the endTransmission  resets the i2c device here:

https://github.com/ambiot/ambpro2_arduino/blob/dev/Arduino_package/hardware/libraries/Wire/src/Wire.cpp

Line: 179 => i2c_reset(((i2c_t *)this->pI2C));

i2c_api.h

    brief  Deinitializes the I2C device 
    param  obj: i2c object define in application software.
    retval none
      
    void i2c_reset(i2c_t *obj);

Because of that the code gets stuck / hangs because the i2c got reset

**MASTER (amb82):**
  
    #include <Wire.h>
    
    #define I2C_SLAVE_ADDR 0x55
    
        void setup() {
        Serial.begin(115200);
        Wire.begin();
    }
    
    void loop() {
        
        int thres = 0;     
        Wire.requestFrom(I2C_SLAVE_ADDR, 1);         
        
        while (Wire.available()) { 
          thres = Wire.read();  
        }     
        delay(10);
        ///////////////////////////
        int tmpval = random(10,100); 
        Wire.beginTransmission(I2C_SLAVE_ADDR); 
        Wire.write(tmpval);            
        Wire.endTransmission(true);   
        delay(10);
    }
  

**SLAVE (ESP32):**
    #include <Wire.h>
    
    #define I2C_SLAVE_ADDR 0x55
    
    void onRequest(){ 
      Wire.write(random(1,10)); 
    }
    
    void onReceive(int len){  
      while(Wire.available()){
        int x = Wire.read();          
        Serial.println(x);  
      }
    }
    
    void setup ( void ) 
    { 
        Wire.setPins(23,19); 
        Wire.begin(I2C_DEV_ADDR );       
          
        Wire.onReceive(onReceive);
        Wire.onRequest(onRequest);
    }
    void loop ( void )
    { }

if i comment line 179 everything works as expect. If you don't think this is a problem please go ahead and close the issue.

Thanks

Kyderio, 2024-01-05:
Hi @moschotto,

I believe the issue does not come from the function endTransmission, but rather from requestFrom function, which is from the master receiving data from esp32. 

In build 4.0.6 early release, we have incorporated the I2C scanner example, which is why we need to do a reset in the endTransmission function. We have tested that I2C MasterReceiveData.ino example is working with Arduino Uno. Maybe you can try increasing the delay?

moschotto, 2024-01-05:
Hi, thanks for the reply. Already tired a delay of 1000ms but this doesn’t change the behavior. Did you try the example code above? 

„Transmitting only“ woks as expected (your example) but if you have a request event as well, it works only once due to the reset event  (i2c_reset in endTransmission). Would it be possible to reproduce it with my code above? Maybe I am wrong…. Please let me know

Thanks

Tomi-ren, 2024-01-16:
I found the same issue, but the difference is that I used both read and write operations in the same loop function, and I found that this reset instruction would cause my read operations to fail. I raised this issue on the forum, but it has not been resolved yet.

Tomi-ren, 2024-01-16:
> Hi, thanks for the reply. Already tired a delay of 1000ms but this doesn’t change the behavior. Did you try the example code above?
> 
> „Transmitting only“ woks as expected (your example) but if you add the request functions as well, it works only once due to the reset event (i2c_reset in endTransmission). Would it be possible to reproduce it with my code above?
> 
> However, if I comment the i2c_reset in endTransmission, my code above (transmit + receive) works as expected, even with a 1ms delay between the receive and transmit function.
> 
> Maybe I am wrong…. Please let me know
> 
> Thanks

hi：look this link：https://forum.amebaiot.com/t/amb82-iic-wire-problem/2593，I seem to have partially solved this problem

Tomi-ren, 2024-02-11:
keep

Kyderio, 2024-02-13:
Hi @moschotto and @Tomi-ren,

Both of you are right, thanks for pointing this out, i2c_reset in Line 179 will be removed and fixed in the next version of the SDK. 

It has already been part of 4.0.7 early release version. You may check this commit. 
( https://github.com/ambiot/ambpro2_arduino/commit/a921adf6ae6165af871893cedac8ace4599cee4e )

Thank you.

moschotto, 2024-02-13:
Thanks for the confirmation

moschotto, 2024-02-13:
.

closed 2024-02-13 by moschotto