Ameba-AIoT/ameba-arduino-pro2 #182: AMB82-mini error on SSD1306-OLED

rkuo2000, 2023-12-25:
[SSD1306_OLED.ino](https://github.com/rkuo2000/Arduino/blob/master/examples/Peripherals/SSD1306_OLED/SSD1306_OLED.ino)
The code was running fine with ESP8266 and ESP32, but failed on AMB82-mini.

In file included from d:\Users\rkuo2\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:13,
                 from D:\Users\rkuo2\Documents\Arduino\examples\Peripherals\SSD1306_OLED\SSD1306_OLED.ino:4:
d:\Users\rkuo2\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:93:41: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SWD_MODE'?
   93 |                      uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);
      |                                         ^~~~~~~~~
      |                                         SWD_MODE
d:\Users\rkuo2\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:102:41: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SWD_MODE'?
  102 |                      uint8_t dataMode = SPI_MODE0);
      |                                         ^~~~~~~~~
      |                                         SWD_MODE

exit status 1

Compilation error: exit status 1

rkuo2000, 2024-02-07:
Yes, it works (using Ameba-Pro2 v4.0.6)
So do my students have to use these customed Adafruit drivers ?  or the fix will be included in v4.0.7 ?

pammyleong, 2024-02-08:
These libraries are third-party libraries and belong to Adafruit, so we are unable to add it to our SDK. It will be added as a third-party library. However, Wire.cpp and Wire.h (I2C) will be using Ameba's one.

M-ichae-l, 2024-02-29:
Close for no further issue.