Ameba-AIoT/ameba-arduino-pro2 #183: VL53L0X running with Error message

rkuo2000, 2023-12-26:
Arduino Library: https://github.com/pololu/vl53l0x-arduino
Arduino Example Code: https://github.com/rkuo2000/Arduino/blob/master/examples/Sensors/VL53L0X/VL53L0X.ino

Finished Compilation and Running :

[ERROR] requestFrom: readed=1, quantity=1 
[ERROR] requestFrom: readed=1, quantity=1 
[ERROR] requestFrom: readed=1, quantity=1 
[ERROR] requestFrom: readed=2, quantity=2 
256 mm

Oliver0804, 2023-12-29:
I recommend using an I2C scanner first to confirm that the Ameba has correctly identified the VL53L0X through I2C.

geofrancis, 2023-12-29:
support have confirmed that the i2c wont work with Arduino libraries but they never explained what the issue was.

Oliver0804, 2023-12-29:
> support have confirmed that the i2c wont work with Arduino libraries but they never explained what the issue was.

They have provided an answer https://github.com/ambiot/ambpro2_arduino/issues/136 , but the revised version has not yet been included in the official release.

rkuo2000, 2023-12-29:
> I recommend using an I2C scanner first to confirm that the Ameba has correctly identified the VL53L0X through I2C.

It found too many devices @@ there is only VL53L0X

I2C Scanner
Scanning...
I2C device found at address 0x01  !
I2C device found at address 0x02  !

Oliver0804, 2023-12-30:
> > I recommend using an I2C scanner first to confirm that the Ameba has correctly identified the VL53L0X through I2C.
> 
> It found too many devices 0x01~0x7E @@ there is only VL53L0X
> 
> I2C Scanner Scanning... I2C device found at address 0x01 ! I2C device found at address 0x02 !

This indicates that I2C is not working properly. https://github.com/ambiot/ambpro2_arduino/issues/136
Im waiting for fix the feature and merge it into the official release.

M-ichae-l, 2024-04-16:
VL53L0X is added at #230

M-ichae-l, 2024-06-20:
close as no further discussion

closed 2024-06-20 by M-ichae-l

geofrancis, 2024-06-20:
Is the amb82 compatible with generic arduino i2c now?