Ameba-AIoT/ameba-arduino-pro2 #186: Can an electronic zoom function be added to achieve automatic focus and autofocus based on some depth of field data?

Tomi-ren, 2023-12-31:
I really hope you can evaluate this feature and look forward to it being implemented. Hold your hands!

Tomi-ren, 2024-01-16:
keep-alive

M-ichae-l, 2024-01-23:
@Tomi-ren 
Thanks to point is out. 
Currently, the camera module is fixed lens and HW limits that it is unable to do automatic focus.

Tomi-ren, 2024-01-29:
I have updated the lens hardware and added lenses that can focus and zoom. I wonder if there is an API interface based on image analysis that can obtain focus judgment, so that I can easily control the motor closed-loop on the focusing and zoom lens. If it cannot be achieved, I speculate that the optical path calculation can be based on the focus and zoom, and a formula can be designed in the control program to achieve open-loop zoom and focusing functions.

Tomi-ren, 2024-02-12:
keep-alive

M-ichae-l, 2024-02-13:
For AMb82-mini the camera module is fixed as JFX37, [datasheet](https://www.amebaiot.com/en/datasheet-download-amb82-mini/). It is possible to change the len for some functions like [IR functions](https://forum.amebaiot.com/t/removing-lens-of-amb86-mini-to-remove-ir-filter-and-placing-an-ir-cut-filter/2599/3).
But for automatic focus, if it requires api update or driver update that is not going to process currently.

M-ichae-l, 2024-04-02:
Close for no further issue

closed 2024-04-02 by M-ichae-l