Ameba-AIoT/ameba-arduino-pro2 #206: U8g2 library compilation error

rkuo2000, 2024-02-08:
### Boards

AMB82-mini

### External Hardware

SSD1306 0.96" OLED

### Hardware Configuration

SSD1306 0.96" OLED

### Version

v4.0.6

### IDE Name

Arduino IDE 2.3.0

### Operating System

Ubuntu 22.04.3 LTS

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

SSD1306 0.96" OLED

### Sketch

```cpp
https://github.com/rkuo2000/Arduino/blob/master/examples/Peripherals/SSD1306_U8g2/SSD1306_U8g2.ino
```

### Error/Debug Message

```plain
In file included from /home/rkuo/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:13,
                 from /home/rkuo/Arduino/examples/Peripherals/SSD1306_U8g2/SSD1306_U8g2.ino:3:
/home/rkuo/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:94:41: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SWD_MODE'?
   94 |                      uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);
      |                                         ^~~~~~~~~
      |                                         SWD_MODE
/home/rkuo/Arduino/libraries/Adafruit_BusIO/Adafruit_SPIDevice.h:103:41: error: 'SPI_MODE0' was not declared in this scope; did you mean 'SWD_MODE'?
  103 |                      uint8_t dataMode = SPI_MODE0);
      |                                         ^~~~~~~~~
      |                                         SWD_MODE

exit status 1

Compilation error: exit status 1
```

### Reproduce remarks

the code runs OK on NodeMCU-32S

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

pammyleong, 2024-02-15:
Hi @rkuo2000 ,

The compilation occurs as different name is define for Ameba and other SPI libraries. Kindly replace the new SPI.h (https://github.com/ambiot/ambpro2_arduino/blob/f506ccbada15c9e0674da15bb0fcf08c545303c8/Arduino_package/hardware/libraries/SPI/src/SPI.h) in the SDK and try again.

If you still encounter any complication errors regarding Adafruit_GFX.h and Adafruit_SSD1306.h, you can use the modified one that is included in the SDK (https://github.com/ambiot/ambpro2_arduino/tree/dev/Arduino_package/hardware/libraries/Wire/src/Adafruit_OLED_libraries).

M-ichae-l, 2024-02-29:
Close as no further issue

closed 2024-02-29 by M-ichae-l