Ameba-AIoT/ameba-arduino-pro2 #210: Playing MP3 file from file system on speaker

ShrivardhanGoenka, 2024-02-17:
### Hardware Environment

AMB82-mini

### Related technical area

Libraries

### Is the feature request related to an un-fixed issue?

No

### Describe the ideal solution/reault/output.

My team wants to play stored MP3 files using the speaker connected to the A out pins. The current multimedia libraries do not support streaming an MP4 which has already ben recorded and stored in the SD card. Alternatively, if possible, we could also use the RTP library to join an RTP broadcast happening in a remote server to play the MP3 file. 

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

h4hassanali, 2024-03-01:
I request for same feature. Please do progress on it ASAP. Thank u

M-ichae-l, 2024-04-16:
Hi all. This process is on the road map. It is going to be added in future versions.

M-ichae-l, 2024-12-06:
Close as feature added in V4.0.8. #277

closed 2024-12-06 by M-ichae-l