Ameba-AIoT/ameba-arduino-pro2 #217: Error in compiling RTSPImageClassification.ino

rkuo2000, 2024-02-24:
### Boards

AMB82-mini

### External Hardware

None

### Hardware Configuration

None

### Version

latest dev (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Ubuntu 22.04.3 LTS

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

Error in compiling RTSPImageClassification

### Sketch

```cpp
examples/AmebaNN/RTSPImageClassification
```

### Error/Debug Message

```plain
/tmp/.arduinoIDE-unsaved2024124-30881-9uix6n.un7gl/RTSPImageClassification/RTSPImageClassification.ino:91:26: error: 'IMAGE_CLASSIFICATION' was not declared in this scope; did you mean 'AUDIO_CLASSIFICATION'?
   91 |     imgclass.modelSelect(IMAGE_CLASSIFICATION, NA_MODEL, NA_MODEL, NA_MODEL, NA_MODEL, DEFAULT_IMGCLASS);
      |                          ^~~~~~~~~~~~~~~~~~~~
      |                          AUDIO_CLASSIFICATION
/tmp/.arduinoIDE-unsaved2024124-30881-9uix6n.un7gl/RTSPImageClassification/RTSPImageClassification.ino:91:88: error: 'DEFAULT_IMGCLASS' was not declared in this scope
   91 |     imgclass.modelSelect(IMAGE_CLASSIFICATION, NA_MODEL, NA_MODEL, NA_MODEL, NA_MODEL, DEFAULT_IMGCLASS);
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

rkuo2000, 2024-02-24:
latest v4.0.7 has the following issues to be patched :
chmod +x /home/rkuo/.arduino15/packages/realtek/tools/ameba_pro2_tools/1.2.17/prebuild_linux
chmod +x /home/rkuo/.arduino15/packages/realtek/tools/ameba_pro2_tools/1.2.17/ino_validation_linux
chmod +x /home/rkuo/.arduino15/packages/realtek/tools/ameba_pro2_tools/1.2.17/nn_model_update_linux

Kyderio, 2024-03-01:
Bug fixed and permission issue solved in latest 4.0.7 build. You can verify and close this issue if it is fixed. Thanks!

closed 2024-03-26 by pammyleong