Ameba-AIoT/ameba-arduino-pro2 #221: Play Audio In headphones from SD card using AMB82 mini

h4hassanali, 2024-03-01:
### Hardware Environment

Amb82 mini

### Related technical area

Audio

### Is the feature request related to an un-fixed issue?

Yes

### Describe the ideal solution/reault/output.

I have an amb82 mini board, I have connected audio jack to DAC and GND Pin and successfully streamed audio from vlc player of my laptop to my headphones connected to audio jack. Now I have an audio file , for example audio.mp3 placed in root directory of my sd card. I want that sd card audio file to play on headphones.

### Additional remarks

I have an amb82 mini board, I have connected audio jack to DAC and GND Pin and successfully streamed audio from vlc player of my laptop to my headphones connected to audio jack. Now I have an audio file , for example audio.mp3 placed in root directory of my sd card. I want that sd card audio file to play on headphones.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

rkuo2000, 2024-03-18:
same request here, I need to play back some sound/voice files stored in SD card.

M-ichae-l, 2024-04-16:
@h4hassanali seems like  this is a duplicate of #210 
Let us close this issue and discuss in #210

closed 2024-04-16 by M-ichae-l