Ameba-AIoT/ameba-arduino-pro2 #229: Neopixel not working

rkuo2000, 2024-03-19:
### Hardware Environment

Adafruit Neopixel

### Related technical area

LED5050 WS2812B

### Is the feature request related to an un-fixed issue?

N/A

### Describe the ideal solution/reault/output.

support Adafruit Neopixel library for Ring, Matrix 

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-05-10:
WS2812B is driven by GPIO with high toggle speed. For this hardware AMB82-mini, we will use SPI clk instead of GPIO. It is under road map will added in future.

rkuo2000, 2024-05-10:
Looking forward to it.

cmlo, 2025-06-27:
This issue has lingered unresolved for an entire year. How much longer must we endure this wait?

M-ichae-l, 2025-07-10:
The [new commit](https://github.com/Ameba-AIoT/ameba-arduino-pro2/commit/0b331774e50f53236c8164edc9ea8451ad7c0d96) had supported it. There will be a new early release this week.

cmlo, 2025-07-10:
Dear M-ichae-I,
I tested your new code for WS2812B LED strip, the result is below:
set TOTAL_LED_NUM=16 is OK
set TOTAL_LED_NUM=8, the index 0 is not working.

治民
---------------------------------------------------------------------
羅治民 博士
數位多媒體設計系副教授
國立台北商業大學(桃園校區)
桃園縣平鎮市福龍路一段100號
電話: 03-4506333   研究室分機: 8746
傳真: 03-4506375
行動電話: 0932-917031
電子郵件: ***@***.***
個人網站: cmlo.tw

cmlo, 2025-07-10:
Dear M-ichae-I,
I tested your new code for WS2812B LED strip, the result is below:
IN SPI0:
set TOTAL_LED_NUM=16 is OK
set TOTAL_LED_NUM=8, the index 0 is not working.
IN SPI1:
set TOTAL_LED_NUM=16 is OK
set TOTAL_LED_NUM=8 is OK

ps. SPI0 also I2C, and I have some devices in I2C port.

治民
---------------------------------------------------------------------
羅治民 博士
數位多媒體設計系副教授
國立台北商業大學(桃園校區)
桃園縣平鎮市福龍路一段100號
電話: 03-4506333   研究室分機: 8746
傳真: 03-4506375
行動電話: 0932-917031
電子郵件: ***@***.***
個人網站: cmlo.tw

M-ichae-l, 2025-07-14:
@cmlo  I have tested your case with this [demo](https://ameba-doc-arduino-sdk.readthedocs-hosted.com/en/latest/ameba_pro2/amb82-mini/Example_Guides/WS2812B/WS2812B%20-%20Patterns.html)

There is no issue for me. Could you try the default example without any change to confirm.