Ameba-AIoT/ameba-arduino-pro2 #231: GPIO port writing.

boburshokirov, 2024-04-05:
### Hardware Environment

AMB82-MINI

### Related technical area

GPIO port writing.

### Is the feature request related to an un-fixed issue?

No.

### Describe the ideal solution/reault/output.

I am trying to connect a TFT display with 8-bit parallel interface. It is working perfectly when I write with digitalWrite(). But the speed is slow. I found a function to write to a GPIO port. But I couldn't make it work. Can you please provide an instructions to use that function?

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

boburshokirov, 2024-04-10:
Can you please be a little bit responsive?

boburshokirov, 2024-04-10:
Is it official repo or should we contact to someone else?

Kyderio, 2024-04-19:
Hi @boburshokirov,

Sorry for the late reply, it may be true that the GPIO speed may not be able to send data fast enough to a TFT display. Can I check what TFT LCD are you using and is there an SPI interface where you can use instead?

Thank you.

boburshokirov, 2024-04-19:
Hi @Kyderio ,

Thanks for the reply. I already tried a couple of other TFT displays with SPI interface. But they are slower than TFT with GPIO. I tried 8-bit parallel writing into the port with other Microcontrollers and it works much faster. I can see micros to write into a port in your library. But I couldn't make it work. If you can provide a simple example of writing into a port, that would be great.

Thanks in advance.

Kyderio, 2024-04-22:
Hi @boburshokirov,

You may refer to the file "wiring_digital.c". There is an API called digitalWrite(uint32_t ulPin, uint32_t ulVal). You may use this API to write into a port. There are many examples in the SDK that uses this API. You may refer to "GPIO->examples->MeasureDistance_HCSR04_Ultrasonic.ino" for reference on how to use this API to write into the port. 

Hope this helps.

Thanks.

boburshokirov, 2024-04-22:
Hi @Kyderio ,

As I mentioned, I want to write into 8 pins at one time. Above example is for writing into 1 pin. digitalWrite(uint32_t ulPin, uint32_t ulVal) is for writing 1 pin, right? If not, can you please provide an example to write into 8 pins at one time.

Thanks

Kyderio, 2024-04-23:
Hi @boburshokirov,

Unfortunately it is not possible at the moment for GPIO 8 bit parallel write. You may consider I2C interface or SPI interface. There will be a delay for writing 8 bits at once. 

Hope this helps.

Thanks.

M-ichae-l, 2024-05-06:
closed as no further discussion.

closed 2024-05-06 by M-ichae-l