Ameba-AIoT/ameba-arduino-pro2 #233: a way to put the red indecator LED out

Reevoy24, 2024-04-15:
### Hardware Environment

AMB82-Mini

### Related technical area

Power LED

### Is the feature request related to an un-fixed issue?

no                                      

### Describe the ideal solution/reault/output.

it is very anoying that the red Power LED is allways on. It would be nice if you could turn the LED off.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

boburshokirov, 2024-04-15:
Don't wait their answer. It's been 2 weeks, I still didn't get any answer. 
Try digitalWrite(24, LOW) in your setup. If it doesn't work, try digitalWrite(23, LOW).

M-ichae-l, 2024-04-16:
@Reevoy24 
Unfortunately Red Power LED, is not controllable by hardware pcb design.

M-ichae-l, 2024-04-22:
close as no further issue

closed 2024-04-22 by M-ichae-l