Ameba-AIoT/ameba-arduino-pro2 #235: SPI library not read output after transfer() call

acrist, 2024-04-21:
### Boards

AMB23/SPI

### External Hardware

28J60 ethernet shield or MCP1251 CAN shield

### Hardware Configuration

Shield connected to SPI pins 12-15 due to official pinout

### Version

v4.0.6

### IDE Name

Arduino IDE 2

### Operating System

macOS

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

I'm trying to connect ethernet shield and modify library to support AMB82 but never had luck. After digging inside library has found that SPI function `transfer()` always return 0, but with description should return response from slave:
Code from library:
`byte SPIClass::transfer(uint8_t _data, SPITransferMode _mode){ // transfer 1 byte data without SS
    (void)_mode;
    
    spi_master_write((spi_t *)pSpiMaster, _data);
    //printf("Master write: %02X\n\r", _data);
    return 0;
}`

Same time, `_mode` has no effect in this point as well.

Description official API says:
`Returns
This function either returns NA or data of 8-bits that transferred through SPI master to the slave.` but from code, it seams that 0 should always returned.

Function `spi_master_write` returning data, has read after data transfer so far I found inside SDK:
(file: component/mbed/targets/hal/rtl8735b/spi_api.c)
`static inline void ssi_write(spi_t *obj, int value)
{
	phal_ssi_adaptor_t phal_ssi_adaptor = &(obj->hal_ssi_adaptor);

	while (!hal_ssi_writable(phal_ssi_adaptor));
	hal_ssi_write(phal_ssi_adaptor, value);
}

static inline int ssi_read(spi_t *obj)
{
	phal_ssi_adaptor_t phal_ssi_adaptor = &(obj->hal_ssi_adaptor);

	while (!hal_ssi_readable(phal_ssi_adaptor));
	return (int)hal_ssi_read(phal_ssi_adaptor);
}

int spi_master_write(spi_t *obj, int value)
{
	ssi_write(obj, value);
	return ssi_read(obj);
}`

Which means that data has been received after transfer should be received.

Meanwhile, I've tried to modify `SPI.cpp` file with some like this:
`byte SPIClass::transfer(uint8_t data, SPITransferMode mode) { // transfer 1 byte data without SS
    (void)mode;
    spi_master_write(pSpiMaster, data);
    u_int8_t _rcvData = spi_master_write(pSpiMaster, data);
    printf("\r\n[INFO] Master write: %02X receive %02X\n", data, _rcvData);
    
    return _rcvData;
}`

and now basic loopback test (jumped SI/SO pins) works, but anyway in general drivers for SPI devices that works not only in "TRANSFER_ONLY" mode, it's not enough because driver stacks at something like:
`17:45:28.529 -> 
[INFO] Master write: 00 receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 00 receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 0A receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 00 receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 00 receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 0A receive FF
17:45:28.529 -> 

17:45:28.529 -> 
[INFO] Master write: 00 receive FF`

Which means that data response not transferred back or it works in some mode, that a'm not familiar yet.

Same time, function `transfer16` will never works well, such as inside method data respect to receiver response from `transfer` function which is always 0:
`uint16_t SPIClass::transfer16(byte pin, uint16_t data, SPITransferMode mode) {
    union {
        uint16_t val;
        struct {
            uint8_t lsb;
            uint8_t msb;
        };
    } t;

    t.val = data;

    if (_bitOrder == LSBFIRST) {
        t.lsb = transfer(pin, t.lsb, SPI_CONTINUE);// <<<< ----- HERE
        t.msb = transfer(pin, t.msb, mode); // <<<< ----- HERE
    } else {
        t.msb = transfer(pin, t.msb, SPI_CONTINUE); // <<<< ----- HERE
        t.lsb = transfer(pin, t.lsb, mode); // <<<< ----- HERE
    }
    //printf("\r\n[INFO] Master write: %04X\n", t.val);

    data = t.val;
    return data;
}`

### Sketch

```cpp
/*
 * A sketch to perform a LOOPBACK
 */ 
#include <SPI.h>

uint16_t caunter;
uint16_t inn;
 
void setup() {
    
    Serial.begin(115200);
    Serial.println("SPI Loopback Test!"); 
   
    SPI.begin();
    delay(10);  
}

void loop(){
  
  for (caunter = 0; caunter <= 1000; caunter++) {
   
    inn = SPI.transfer(caunter,SPI_CONTINUE);
  
     Serial.print (caunter, DEC);
     Serial.print(" -- ");
     Serial.println(inn, HEX);
    
delay(500);

   }
}
```

### Error/Debug Message

```plain
18:27:59.405 -> store fcs data for application 
18:27:59.405 -> RAM TM_STATUS 0x00bf1208 err 0x00001208
18:27:59.405 ->  read fcs_status 0x000000bf
18:27:59.438 ->  read fcs_status 0x000000bf
18:27:59.438 -> === Process VOE IMG ===
18:27:59.438 -> [Image Start Table @ 0x20106200]
18:27:59.438 -> RAM Load @ 0x80fd100->0x20106200, 0x5c48
18:27:59.438 -> DDR Load @ 0x8103080->0x70100000, 0x180e4
18:27:59.438 -> === FW Load Done ===
18:27:59.438 -> 
18:27:59.438 -> Boot Loader <==
18:27:59.438 -> 
18:27:59.438 -> == RAM Start ==
18:27:59.470 -> Build @ 17:05:27, Dec 20 2023
18:27:59.470 -> 
18:27:59.470 -> $8735b>SPI Loopback Test!
18:27:59.470 -> 0 -- 0
18:27:59.963 -> 1 -- 0
18:28:00.457 -> 2 -- 0
18:28:00.986 -> 3 -- 0
18:28:01.477 -> 4 -- 0
18:28:01.966 -> 5 -- 0
18:28:02.456 -> 6 -- 0
18:28:02.983 -> 7 -- 0
18:28:03.474 -> 8 -- 0
18:28:03.967 -> 9 -- 0
18:28:04.458 -> 10 -- 0
18:28:04.984 -> 11 -- 0
18:28:05.481 -> 12 -- 0
18:28:05.972 -> 13 -- 0
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

M-ichae-l, 2024-05-10:
the board hardware is 
> Boards
> AMB23/SPI

Could I double check the hardware?

acrist, 2024-05-10:
Board is AMB82-MINI