Ameba-AIoT/ameba-arduino-pro2 #247: AMB82 mini - adding a new camera sensor

kbayat, 2024-06-19:
### Hardware Environment

AMB82 mini

### Related technical area

adding support for a new camera sensor

### Is the feature request related to an un-fixed issue?

No

### Describe the ideal solution/reault/output.

If possible, please provide  support for the image sensor  OV02B10:

https://www.digikey.com/en/products/detail/omnivision-technologies-inc/OV02B10-A25A-001A/14622531
  (with 7-bit I2C address of 0x3C)

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-06-20:
Please refer to the follow file, 
https://github.com/ambiot/ambpro2_arduino/blob/aae61c269b61d764c2ea4f42a1a0651c0ebcd1fa/Arduino_package/hardware/system/project/realtek_amebapro2_v0_example/inc/sensor.h#L94

Currently, the SDK only supports F37 and GC5035. There is no limitation to supports the rest of the sensors by updating SDK. 
However, if the sensor beyond the following, that has limits to be supported. 
> #define SENSOR_SC2336       0x01
> #define SENSOR_GC2053       0x02
> #define SENSOR_GC4653 		0x03
> #define SENSOR_F37			0x04
> #define SENSOR_IMX327		0x05
> #define SENSOR_F51			0x06
> #define SENSOR_PS5258       0x07 //It don't support the multi sensor for PS5258 now.If you want to use the sensor,please remove it.
> #define SENSOR_SC301		0x08
> #define SENSOR_IMX307		0x09
> #define SENSOR_SC2333		0x0A
> #define SENSOR_GC4023		0x0B
> #define SENSOR_PS5420		0x0C
> #define SENSOR_PS5270		0x0D
> #define SENSOR_GC5035		0x0E
> #define SENSOR_PS5268		0x0F
> #define SENSOR_SC2310		0x10
> #define SENSOR_PS5420_HDR	0x11
> #define SENSOR_PS5270_HDR	0x12
> #define SENSOR_F53			0x13
> #define SENSOR_F55			0x14
> #define SENSOR_GC4663		0x15
> #define SENSOR_GC4663_HDR	0x16
> #define SENSOR_K351			0x17
> #define SENSOR_K351_HDR		0x18
> #define SENSOR_OV50A40		0x19
> #define SENSOR_SC301_HDR	0x1A

M-ichae-l, 2024-07-03:
close as no further discussion

closed 2024-07-03 by M-ichae-l