Ameba-AIoT/ameba-arduino-pro2 #251: ControlLED abort execution (using v4.0.8-build20240703)

rkuo2000, 2024-07-08:
### Boards

AMB82-mini

### External Hardware

None

### Hardware Configuration

None

### Version

latest dev (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Ubuntu 22.04.3 LTS

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

WiFi / simpleHttpWeb / ControlLED
ControlLED.ino abort execution 

### Sketch

```cpp
none (it's example code)
```

### Error/Debug Message

```plain
=== Process ISP_IQ ===
=== Load Done ===
=== Load BL ===
[Image Start Table @ 0x18200]
=== Load Done ===

== Boot Loader ==
Jun 28 2024:17:18:16
=== Load FCS Para ===
=== Load Done ===
[crc pass]
=== Load ISP_IQ Sensor ===
ISP_IQ @ 0x8461080, 0x1af80
=== Process ISP_IQ ===
=== Load Done ===
=== Load FW1 ===
FW_ISP_IQ @ 0x8061080, 0x1cf80
=== Process FW_ISP_IQ ===
DRAM_TYPE is DDR2 128MB.
ddr_freq = 533
VOE flash @ 0x807e080, 0x7ff80
FCS KM_status 0x00000082 err 0x00000000
Wait KM fcs done 0 us
FCS TM_status 0x00000001 
dark mode 0 adc final value 0 sample 1 mode 0
fcs final cmd 0x3d7a0b48 index 17
store fcs data for application 
fcs OK
Can't find ff ff ff ff
load time sensor:254us iq:4587us itcm:1011us dtcm:51us ddr:11957us ddr2:11310us
=== Process VOE IMG ===
[Image Start Table @ 0x20106200]
RAM Load @ 0x80fe100->0x20106200, 0x5edc
DDR Load @ 0x8104080->0x70100000, 0x13647c
=== FW Load Done ===

Boot Loader <==

== RAM Start ==
Build @ 17:22:43, Jun 28 2024

$8735b>76dfffe0 exceed 75c00000
It can't be allocate buffer

abort execution
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

rkuo2000, 2024-07-08:
this example code works fine if compiled by 4.0.7-build20240610

aramix, 2024-07-09:
Having the same issue, with any code. Had to downgrade to 4.0.7 again

M-ichae-l, 2024-07-10:
Thanks for the feedback. 

Note the "build" version SDK is pre-release version. It may not stable and have issues. 
For stable SDK please refer to released version SDK. https://github.com/ambiot/ambpro2_arduino/releases

pammyleong, 2025-02-14:
Tested on latest pre-release version 4.0.9-build20250205 and ControlLED example working fine on ubuntu.