Ameba-AIoT/ameba-arduino-pro2 #253: I hope the online model converter can support ONNX model conversion, not just Darknet models.

Oliver0804, 2024-07-13:
### Hardware Environment

AMB82mini

### Related technical area

NN

### Is the feature request related to an un-fixed issue?

I hope the online model converter can support ONNX model conversion, not just Darknet models.

### Describe the ideal solution/reault/output.

.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-12-06:
https://www.amebaiot.com/en/amebapro2-ai-convert-model/

I believe the onnx is already supported by some types of model.

M-ichae-l, 2025-01-13:
Close as no further issue.

closed 2025-01-13 by M-ichae-l