Ameba-AIoT/ameba-arduino-pro2 #263: Please add support for IMX327 Sensor

svjaa, 2024-09-11:
### Hardware Environment

AMB82 Mini

### Related technical area

add sensors

### Is the feature request related to an un-fixed issue?

Please add support for the IMX327 sensor as listed in [this ](https://forum.amebaiot.com/t/amb82-mini-adding-a-new-camera-sensor/3029)thread on the official forums.

Thank you.

### Describe the ideal solution/reault/output.

When this request is implemented, the IMX327 sensor will be supported for use by the Arduino SDK.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-10-04:
Hi @svjaa the latest build version is supported the sensor. Please refer to [4.0.8-build20241003](https://github.com/Ameba-AIoT/ameba-arduino-pro2/releases/tag/V4.0.8-QC-V05)

ryanstout, 2024-10-26:
@svjaa mind if I ask how you sourced the imx327? Does anyone ship the sensor with a 24 pin compatible cable? If not, I’m assuming you had to add some regulators to get the right voltages? Any help duplicating your setup would be greatly appreciated. Thanks!

M-ichae-l, 2024-12-06:
Close as feature already supported.

closed 2024-12-06 by M-ichae-l