Ameba-AIoT/ameba-arduino-pro2 #264: SD MMC

peder2key, 2024-09-16:
### Hardware Environment

AMB82-MINI

### Related technical area

Add possibility to get USB - SD MMC

### Is the feature request related to an un-fixed issue?

Not sure. 

### Describe the ideal solution/reault/output.

When using USB-OTG on a computer it should act like a SD card. So that the user don't need to take out the SD card to see the video or copy it. 

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

pammyleong, 2025-03-17:
Please check USB Mass Storage example.

https://github.com/Ameba-AIoT/ameba-arduino-pro2/blob/dev/Arduino_package/hardware/libraries/USB/examples/USB_Mass_Storage/USB_Mass_Storage.ino

 Thank you.

Emidizda, 2025-03-17:
Hello @pammyleong,

I tried compiling this to the AMB82-MINI and it is failing. Will this work for the AMB82-MINI?

Best regards

pammyleong, 2025-03-17:
Yes, its for AMB82 Mini. May I know if you downloaded the relevant cpp and header file as well?

https://github.com/Ameba-AIoT/ameba-arduino-pro2/pull/312/files (Can find the files from here)

As we have not release a pre-release version, you have to manually download and add to your SDK.

Emidizda, 2025-03-17:
Aha, my fault. After adding the source files everything works 💯 

Great work, thank you! 

Best regards

Emidizda, 2025-03-18:
Hello @pammyleong,

I have been playing around with this a little, but I am not able to get the correct status when I plug in the usb cable: 

#include "USBMassStorage.h"

USBMassStorage USBMS;

void setup() {
  Serial.begin(115200);
  USBMS.USBInit();
  USBMS.SDIOInit();

  int status = USBMS.USBStatus();
  Serial.print("Status ");
  Serial.println(status);

  USBMS.initializeDisk();
  USBMS.loadUSBMassStorageDriver();

  delay(5000);
}

void loop() {
  int status = USBMS.USBStatus();
  Serial.print("Status ");
  Serial.println(status);
   delay(1000);
}

The status is always 0, even if the USB cable is not connected. Any ideas?

Best regards

pammyleong, 2025-03-19:
Hi @Emidizda, 

Apologies for the confusion, I will remove this print log as this USBStatus function is actually only checking on the USB Init status and not if the USB is attached.

[image]

Emidizda, 2025-03-19:
Hello @pammyleong 

But I will like to know the status if the usb is connected or not. How can I figure that out?

Best regards

pammyleong, 2025-04-04:
Hi @Emidizda ,

Sorry for the late reply.

You may refer to my Pull Request.

https://github.com/Ameba-AIoT/ameba-arduino-pro2/pull/314/files

You can use isConnected API to get the connection status.

Emidizda, 2025-05-07:
Hello @pammyleong,

I have been playing around with the SD MMC, but after a while I am getting this error: 

== RAM Start ==
Build @ 14:04:33, Mar 25 2025

$8735b>OTG DRIVER VER: 0826

USB init status 0

SD_Init 0

sector number: 122138624

capacity: 59638 MB

USB MSC driver load done, Available heap [0x5a9b5e0]

SD_DeInit

SD_Init 0

sd_reset_procedure ret = 0

usbd_msc_bulk_in_complete: ESHUTDOWN status = -108

usbd_msc_bulk_in_complete: ESHUTDOWN status = -108

usbd_msc_bulk_out_complete: tx endpoint shutdown, code -108

Do you have any information of why I am getting this? I am not able to see the SD card on the computer anymore either. But if I take it out at plug it directly to the computer the SD card works just fine. 

Best Regards

Emidizda, 2025-07-01:
I have had some time to test now and it seems to be working better now. Have you changed something?

pammyleong, 2025-07-08:
No, we didnt change anything