Ameba-AIoT/ameba-arduino-pro2 #268: OTA Issues in Pre Release Version 4.0.8 - 2024/10/03

TeaBear5, 2024-10-03:
### Boards

AMB82-MINI

### External Hardware

USB Connection to PC

### Hardware Configuration

None

### Version

latest dev (checkout manually)

### IDE Name

Arduino IDE 2.3.3 & arduino-cli 1.0.0-4c.1 Commit: ba19a2d3

### Operating System

Ubuntu 22.04.4 LTS

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

I am having issues using the [ameba-OTA-UI](https://github.com/Ameba-AIoT/ameba-OTA-UI) to properly run OTA when linked to the example OTA project in the following dev release 4.0.8-build20241003.

Frist I changed the ip address in the ota_drv.c file shown below:
[image]

That made me able to get my device to show up in the UI as shown below:
[image]

However, i can't seem to get the OTA to trigger using the UI.  to further clarify the following is the response on the device side:
```
[http_update_ota] Download new firmware begin, total size : 5447680

[http_update_ota] Current firmware index is 1
..
[ota_flash_NOR] target_fw_addr=0x520000, target_fw_len=0x10000

[http_update_ota] ota flash failed
```
On the UI side the folloing request seems to error out:
[image]

for further context the connectedclients request seems to be working properly:
```
{
    "recentEntries": [
        {
            "ip": "192.168.30.45",
            "timestamp": 1727979457117,
            "OTA_state": "OTA_STATE_DOWNLOAD_FIRMWARE_COMPLETED"
        }
    ]
}
```

But the start_ota request seems to have an issue in getting a proper response:
[image]

Any help would be appreciated. I plan to take a deeper look at the OTA drivers, but was wondering if there was something obvious i was missing before doing so.

### Sketch

```cpp
#include "ota_thread.h"
#include "WiFi.h"

char ssid[] = "Network_SSID";    // your network SSID (name)
char pass[] = "Password";        // your network password
int status = WL_IDLE_STATUS;

void setup()
{
    Serial.begin(115200);

    // Connection to internet
    while (status != WL_CONNECTED) {
        Serial.print("Attempting to connect to WPA SSID: ");
        Serial.println(ssid);
        status = WiFi.begin(ssid, pass);
        delay(2000);
    }

    // Set up the threads
    start_OTA_threads();
}

void loop()
{
    // Empty or add non-blocking code here
}
```

### Error/Debug Message

```plain
See Description (No Compile or upload error).
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

kev0077, 2024-10-04:
hi @TeaBear5 , I think you have missed out the following steps in creating the OTA firmware image.

1. Set NN Model Load From SD Card. “Tools” -> “NN Model Load From” -> “SD Card”
[image]
2. Enable the OTA mode. “Tools” -> “OTA Mode” -> “Enable”
[image]

Compile the code again using Arduino IDE, you will then find the ``ota.bin`` file in the /home/username/.arduino15/packages/Realtek/tools/ameba_d_tools/x.x.x/ folder 
[image]
This is the .bin file to be uploaded to your UI server for OTA trigger.

Thank you.

TeaBear5, 2024-10-04:
@kev0077 Thanks!  This works great!

For anyone else's reference this can also be accomplished using the arduino-cli tool with a command similar to the following:
```
arduino-cli compile --fqbn realtek:AmebaPro2:Ameba_AMB82-MINI --output-dir ./build \
--build-property build.ota_mode_val=Enable \
--build-property build.ota_mode_flags=-DArduino_OTA_MODE \
--build-property build.Arduino_LOAD_MODEL_flags=-DArduino_LOAD_MODEL_SD \
--build-property build.model_src=LoadFromSD \
OTA_Test
```

closed 2024-10-04 by TeaBear5