Ameba-AIoT/ameba-arduino-pro2 #279: WiFiClient not working as expected

trevorwslee, 2024-11-08:
### Boards

AMB82-mini

### External Hardware

no`

### Hardware Configuration

no

### Version

other

### IDE Name

Arduino IDE 2.3.3

### Operating System

Windows

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

The sample SimpleTCPServer.ino (https://github.com/Ameba-AIoT/ameba-arduino-pro2/blob/dev/Arduino_package/hardware/libraries/WiFi/examples/SimpleTCPServer/SimpleTCPServer.ino) works fine.

However, if the loop() is changed to be like

```
void loop()
{
    WiFiClient client = server.available();
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.print(c);
      }
    }
    client.stop();
}

```
it doesn't seem to work as expected,

Note that I tried the same modified sketch with UNO R4 and it works fine with UNO R4

Thanks for looking into the issue.

### Sketch

```cpp
#include <WiFi.h>

char ssid[] = "xxx";    // your network SSID (name)
char pass[] = "xxx";        // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                // your network key Index number (needed only for WEP)
int status = WL_IDLE_STATUS;     // Indicator of Wifi status

WiFiServer server(5000);

void setup()
{
    Serial.begin(115200);    // initialize serial communication

    // attempt to connect to Wifi network:
    while (status != WL_CONNECTED) {
        Serial.print("Attempting to connect to Network named: ");
        Serial.println(ssid);    // print the network name (SSID);

        // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
        status = WiFi.begin(ssid, pass);
        // wait 10 seconds for connection:
        delay(10000);
    }
    //server.setNonBlockingMode();
    server.begin();       // start the tcp server on port 5000
    printWifiStatus();    // you're connected now, so print out the status
}

const size_t bufferLen = 1;
char buffer[bufferLen];
void loop()
{
    WiFiClient client = server.available();
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.print(c);
      }
    }
    client.stop();
}

void printWifiStatus()
{
    // print the SSID of the network you're attached to:
    Serial.print("SSID: ");
    Serial.println(WiFi.SSID());

    // print your WiFi IP address:
    IPAddress ip = WiFi.localIP();
    Serial.print("IP Address: ");
    Serial.println(ip);

    // print the received signal strength:
    long rssi = WiFi.RSSI();
    Serial.print("signal strength (RSSI):");
    Serial.print(rssi);
    Serial.println(" dBm");
}
```

### Error/Debug Message

```plain
none
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

Kyderio, 2024-11-22:
Hi, have you tried setting non blocking mode for your server? Uncomment this line: //server.setNonBlockingMode(); in your set up loop. Thank you.

trevorwslee, 2024-11-23:
If I enable non-blocking mode, I will get error like:

[ERROR] get_available Accept connection failed

right after the sketch started looping in loop()

and here is the sketch I used

/*

 Example guide:
 https://www.amebaiot.com/en/amebapro2-arduino-server-client/
 */

#include <WiFi.h>

char ssid[] = "...";    // your network SSID (name)
char pass[] = "...";        // your network password (use for WPA, or use
as key for WEP)
int keyIndex = 0;                // your network key Index number (needed
only for WEP)
int status = WL_IDLE_STATUS;     // Indicator of Wifi status

WiFiServer server(5000);

void setup()
{
    Serial.begin(115200);    // initialize serial communication

    // attempt to connect to Wifi network:
    while (status != WL_CONNECTED) {
        Serial.print("Attempting to connect to Network named: ");
        Serial.println(ssid);    // print the network name (SSID);

        // Connect to WPA/WPA2 network. Change this line if using open or
WEP network:
        status = WiFi.begin(ssid, pass);
        // wait 10 seconds for connection:
        //delay(10000);
        for (int i = 0; i < 10; i++) {
          Serial.println("... wait ...");
          delay(1000);
        }
    }
    server.setNonBlockingMode();  // R4 doesn't have this
    server.begin();       // start the tcp server on port 5000
}

void loop()
{
    WiFiClient client = server.available();
    bool firstConnected = false;
    while (client.connected()) {
      if (firstConnected) {
        Serial.println("***** connected *****");
      }
      firstConnected = true;
      if (client.available()) {
        char c = client.read();
        Serial.print(c);
      }
    }
    client.stop();
    Serial.println("***** the end *****");
}

closed 2026-02-02 by M-ichae-l