Ameba-AIoT/ameba-arduino-pro2 #280: help implement object detection for single image on SD card

NFUEEchris, 2024-11-08:
### Hardware Environment

AMB82-mini

### Related technical area

object detection

### Is the feature request related to an un-fixed issue?

plz upgrade object detection for single image on SD card first

### Describe the ideal solution/reault/output.

plz upgrade object detection for single image on SD card first

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

M-ichae-l, 2024-12-06:
Could you explain more on the application you request? 
What is "single image on SD card first" by details? 

Do you request an object detection on image instead of streaming?

NFUEEchris, 2024-12-06:
> Do you request an object detection on image instead of streaming?
yes, if it's possible, thank you.

pammyleong, 2025-02-14:
Hi @NFUEEchris,

You can make use of AmebaFatFS to save your image on an SD card.

Kindly refer to the following example.

[ObjectDetectionJPEGSDCard.zip](https://github.com/user-attachments/files/18793215/ObjectDetectionJPEGSDCard.zip)

- You may create another channel for taking snapshots. And you have to configure the channel as follows,
 VideoSetting configJPEG(VIDEO_FHD, CAM_FPS, VIDEO_JPEG, 1); with the last argument set as 1, which means enable snapshot.
- If you would like your image to have OSD overlay, you will have to draw on CHANNELJPEG as well (illustrated in the reference example).

closed 2025-02-17 by M-ichae-l

NFUEEchris, 2025-02-17:
sorry, this is not what i wanted, i want to detect avaliable data(image) in SD card

pammyleong, 2025-02-18:
Hi @NFUEEchris ，

We have an example that will do object detection for images found on the SD Card. However, this example will only output detected objects to a JSON file. If you want to draw a bounding box, you may need to explore and modify this example. The example guide is currently in progress. 

Example: https://github.com/Ameba-AIoT/ameba-arduino-pro2/blob/dev/Arduino_package/hardware/libraries/NeuralNetwork/examples/ObjectDetectionImage/ObjectDetectionImage.ino

You can see if this is what you are looking for.

NFUEEchris, 2025-02-19:
thx you gay did great job!!

PR #300 (2025-02-17): Add ObjectDetection result saving as MP4 file or JPEG example
## Description of Change
Add ObjectDetection result saving as MP4 file or JPEG example

## Tests and Environments 
- AMB82-mini
- ambpro2_arduino V4.0.9
- Arduino IDE 2.3.4
- Windows 10

## Remark related links
- Closes #280