Ameba-AIoT/ameba-arduino-pro2 #282: I2C does not send read when Wire.requestFrom(...)

eLEcTRiCZiTy, 2024-11-25:
### Boards

AMB82-mini

### External Hardware

Saleae Logic8 attached to pin 12, pin 13 (I2C).

### Hardware Configuration

satel-vl53l7cx modified to bypass onboard regulator, so SB1 short and SB2 open, then AVDD=IOVDD (sensor pin - AMB82-mini pin):
- SDA - pin 12
- SCL - pin 13
- GND - GND
- IOVDD - VDD33

(Note: fully functional and tested with a few STM32 boards, ESP32, and similar)

### Version

v4.0.8

### IDE Name

Arduino IDE 2.3.3

### Operating System

WIndows 10, Windows 11, Ubuntu 22.04

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

When I checked If sensor initializes properly, I found, that I2C Write is OK but I2C Read writes only the address but never start read. The read part is totally missing. I do not have other I2C readable sensors that have simple code. So I will not provide code sample to reproduce, but it should be any read attempt. 

[image]

### Sketch

```cpp
code of functions I will use:

typedef struct
{
  TwoWire *port; //== &Wire
  uint16_t address; //== 0x52
} VL53L7CX_Platform;

uint8_t VL53L7CX_RdByte(
  VL53L7CX_Platform *p_platform,
  uint16_t register_address,
  uint8_t *p_value)
{
  uint8_t status = 0;	
  p_platform->port->beginTransmission(p_platform->address>>1);
  p_platform->port->write(highByte(register_address));
  p_platform->port->write(lowByte(register_address));
  p_platform->port->endTransmission();
  p_platform->port->requestFrom(p_platform->address>>1, 1U);
  *p_value = p_platform->port->read();
return status;
}

uint8_t VL53L7CX_WrByte(
  VL53L7CX_Platform *p_platform,
  uint16_t register_address,
  uint8_t value)
{
  uint8_t status = 0;
  p_platform->port->beginTransmission(p_platform->address>>1);
  p_platform->port->write(highByte(register_address));
  p_platform->port->write(lowByte(register_address));
  p_platform->port->write(value);
  status =  p_platform->port->endTransmission();
  return status;
}

Initialization send these messages:

...
/* SW reboot sequence */
status |= VL53L7CX_WrByte(&(p_dev->platform), 0x7fff, 0x00);
status |= VL53L7CX_WrByte(&(p_dev->platform), 0x0009, 0x04);
status |= VL53L7CX_WrByte(&(p_dev->platform), 0x000F, 0x40);
status |= VL53L7CX_WrByte(&(p_dev->platform), 0x000A, 0x03);
status |= VL53L7CX_RdByte(&(p_dev->platform), 0x7FFF, &tmp);
status |= VL53L7CX_WrByte(&(p_dev->platform), 0x000C, 0x01);
...

As seen on the logic analyzator image, there is Write(index,data) [ADDR(W), 0x00, 0x01, 0x03, STOP], then Write(index)/Read(data) [ADDR(W), 0x7F, 0xFF] but missing Read(data) part which should be [ADDR+1(R), DATA]. It continues with another Write(index, data) [ADDR, 0x00, 0x0C, 0x01].

There is expected missing Read part where data is received from slave buffer.
```

### Error/Debug Message

```plain
No error messages. Only boot-up and my printouts:

== Rtl8735b IoT Platform ==

[Normal mode]
BootFromNORFlash
[Start Boot ROM...]
=== Load PARTBL ===
=== Load Done ===
=== Load ISP_IQ ===
[fcs chk pass]
ISP_IQ @ 0x8461080, 0x14f80, 0x0
mfcs_data version 0x00010001
fcs_data version 0x00010101
=== Process ISP_IQ ===
=== Load Done ===
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [X] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

eLEcTRiCZiTy, 2024-11-25:
I tried various implementation of the same Write and Read/Write functions and all do the same output on logic analyzer. The Read section is totally missing.

Kyderio, 2024-12-02:
Hi @eLEcTRiCZiTy,

Please add a delay right after the Wire.requestFrom(...) API.

Thank you.

eLEcTRiCZiTy, 2025-02-09:
Hi, I trying to catch and diagnostic another issue with I2C and I do not know it this is related to my another problem.

In case of this issue, the delay really helped and solved the issue. The Read method do not check if I2C interface is ready to read.

closed 2025-02-09 by eLEcTRiCZiTy