Ameba-AIoT/ameba-arduino-pro2 #284: Streaming RTSP with a recorded mp4 from SD card

peder2key, 2024-11-30:
### Hardware Environment

AMB82-MINI

### Related technical area

Wifi, SD and camera

### Is the feature request related to an un-fixed issue?

No

### Describe the ideal solution/reault/output.

Stream selected files from SD card. This would be a nice feature if you would like to look at the video without copying the video over from SD card to a computer. It could also be nice to have a way of looking at a video without taking out the SD card.

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

eLEcTRiCZiTy, 2024-12-01:
Hi, 
I think you can program it by yourself. You have all needed parts: RTSP streaming example with MP4 recording, SD card example. You just have to use your arduino programming skills to do the rest by yourself.

I do not think Realtek Ameba team would implement each feature combination as example.

peder2key, 2024-12-02:
I have taken a look at those examples, but I am not able to see a way of replaying a video. From the amebaFatFSFile there is no obvious way of streaming back the file. There is a method called convertMp3ToArray that seems to convert a mp3 to a sound. But do you have a idea? Seems like the streamIO expects a MMFModule that I am not sure how to get from a amebaFatFs.

Kyderio, 2024-12-04:
Hi @peder2key, 

If you mean doing RTSP stream while recording the video, it is not possible. However if you mean to do RTSP stream with a recorded MP4 recording. Thee example is there.

Thank you.

peder2key, 2024-12-05:
Hi @Kyderio 

Could you please give me a hint on which examples shows the possibility to stream a mp4 file from SD with RTSP?

Best Regards

pammyleong, 2025-02-18:
Hi @Emidizda,

I would like to understand your need for this request.

Do you mean that you would like a feature whereby AmebaPro2 Mini acts like a "card reader" whereby you can access the MP4 files in the SD card without removing it from the board?

Could you elaborate more on what do you mean by "Streaming RTSP with a recorded mp4 from SD card"? Can I understand it by you would like to play and watch the MP4 file after recording without removing the SD card? Thank you!

pammyleong, 2025-03-07:
Hi @Emidizda ，

Kindly try out the Demuxer RTSP example in the pre-release version 4.0.9-build20250303.

Example Guide: https://ameba-doc-arduino-sdk.readthedocs-hosted.com/en/latest/amebapro2/Example_Guides/Multimedia/RTSP%20Demuxer.html

pammyleong, 2025-03-17:
Issue closed as no further requests or issues.

closed 2025-03-17 by pammyleong