Ameba-AIoT/ameba-arduino-pro2 #287: RTSP Stream dec profile in AMB82-mini

ManikandanVaradharaj, 2024-12-05:
### Hardware Environment

AMB82-mini

### Related technical area

H264 enc/dec

### Is the feature request related to an un-fixed issue?

I'm using AMB82-mini camera as rtsp server,Here my display module will only support h264 dec in baseline profile,so that i couldn't stream the video in the display.

### Describe the ideal solution/reault/output.

00 00 00 01 27 64 00 33 ac 13 1a a0 78 02 27 e5

64: Profile IDC
0x64 corresponds to the High profile.
00: Constraint Set Flags

Constraint flags are all 0 in this case, meaning no special constraints are applied.
33: Level IDC

0x33 corresponds to Level 5.1, meaning the decoder must support high-performance video streams (e.g., up to 4K resolution at high frame rates).

### I have checked existing/closed list of issues and the Contributing Guide.

- [X] I confirm I have checked existing/closed list of issues and the Contributing Guide.

ManikandanVaradharaj, 2024-12-10:
After i changed params in video_drv.c file .level = VCENC_H264_LEVEL_1,
    .profile = VCENC_H264_BASE_PROFILE,
    .cavlc = 0, In log I'm getting error as [VOE]VCEncSetCodingCtrl: ERROR Invalid entropy coding mode for baseline profile[VOE]VCEncSetCodingCtr[VOE]frame_el() failed.nd: sensor d[VOE]ENC opidn't initiaen fail
lize doha !v

_send2voe too long 172322 cmd 0x00000206 p1 0x00000000 p2 0x00000000

VOE command 0x206 fail ret 0x1

VOE_OPEN_CMD command fail
hal_video_open fail

[VID Err]Please check sensor id first,the id is 1

[VOE]VCEncStrmEncode: ERROR Null argument
[VOE]ENC Fail ret -2

kev0077, 2024-12-12:
Hello, as per our forum discussion, I have suggested you to configure `video_params` structure in cores/ambpro2/video_drv.c 
[image]
and use `VideoSetting config(VIDEO_VGA, CAM_FPS, VIDEO_H264, 0);` instead of `VideoSetting config(CHANNEL);`, I believe you are able to view the RTSP stream now after making the changes. 
Regarding the 5s delay on streaming, you might need to check whether there is any delay imposed on the streaming viewer side, for example, in VLC player, you can reduce the caching time before opening the streaming. Besides, network speed and connectivity could be the cause for the delay too. Thank you.

Sindhuja-15, 2024-12-26:
@kev0077 Can you tell me how can i send the rtsp stream video/audio data to specific tcp server. In example it sends the data in local wifi ip.