Ameba-AIoT/ameba-arduino-pro2 #298: AMB82-mini is not able to communicate with STM vl53l5cx or vl53l7cx using Sparkfun library based on official library/driver

eLEcTRiCZiTy, 2025-02-09:
### Boards

AMB82-mini

### External Hardware

Plain AMB82-mini devkit with default camera attached.

### Hardware Configuration

STM VL53L7CX eval. kit or Sparkfun ToF Imager with VL53L5CX or Pololu version of VL53L5CX or VL53L7CX attached with jumper wires. Sensors have the same internals and are fully compatible.
AMB82-mini -> VL53L7CX (Sparkfun )
VDD33 -> 3V3
GPIO 12 -> SDA
GPIO 13 -> SCL
GND -> GND

### Version

v4.0.8

### IDE Name

Arduino IDE 2.3.3

### Operating System

WIndows 11

### Auto Flash Mode

Disable

### Erase All Flash Memory (16MB)

Disable

### Standard Lib

Arduino_STD_PRINTF

### Upload Speed

2000000

### Description

Install the library from Sparkfun - Sparkfun VL53L5CX Arduino Library in IDE or download it from ([SparkFun_VL53L5CX_Arduino_Library](https://github.com/sparkfun/SparkFun_VL53L5CX_Arduino_Library)).

Tested also with modified library ([SparkFun_VL53L5CX_IO.cpp](https://github.com/sparkfun/SparkFun_VL53L5CX_Arduino_Library/blob/main/src/SparkFun_VL53L5CX_IO.cpp) ) with added small delay after Wire.requestFrom(...) and before .read(...) according to #282. I tested delay(1) and delay(10) with the same effect.

Try to use any of examples, the result is the same printout in serial monitor. I tried to wait for 5 minutes without any change or continuation.

I implemented Arduino library myself at first based on library sources from ST Microelectronics for STM32 MCUs. I tried various implementations of the adapter (platform.h, platform.c). The added delay from #282 helped and I was able to read responses at the start of the initialization. The issue is the uploading of the sensor firmware I think. The very begining of the sending of the large portion of image is sent correctly and my new new logic analyzer (Analog Discovery 3) was able to decode the packets, but after short while the sent i2c packets looked somewhat OK, but AD3 wasn't able to decode them. I do not have other logic analyzer or mixed data oscilloscope at hand to verify, where is the problem. So There might be issue in sending of larger bulk of the data using I2C. I moved to Sparkfun arduino library because it worked with ESP32 out of the box and it is publicly available.

### Sketch

```cpp
/*
  Read an 8x8 array of distances from the VL53L5CX
  By: Nathan Seidle
  SparkFun Electronics
  Date: October 26, 2021
  License: MIT. See license file for more information but you can
  basically do whatever you want with this code.

  This example shows how to read all 64 distance readings at once.

  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/18642

*/

#include <Wire.h>

#include <SparkFun_VL53L5CX_Library.h> //http://librarymanager/All#SparkFun_VL53L5CX

SparkFun_VL53L5CX myImager;
VL53L5CX_ResultsData measurementData; // Result data class structure, 1356 byes of RAM

int imageResolution = 0; //Used to pretty print output
int imageWidth = 0; //Used to pretty print output

void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("SparkFun VL53L5CX Imager Example");

  Wire.begin(); //This resets to 100kHz I2C
  Wire.setClock(400000);
  
  Serial.println("Initializing sensor board. This can take up to 10s. Please wait.");
  if (myImager.begin() == false)
  {
    Serial.println(F("Sensor not found - check your wiring. Freezing"));
    while (1) ;
  }
  
  myImager.setResolution(8*8); //Enable all 64 pads
  
  imageResolution = myImager.getResolution(); //Query sensor for current resolution - either 4x4 or 8x8
  imageWidth = sqrt(imageResolution); //Calculate printing width

  myImager.startRanging();
}

void loop()
{
  //Poll sensor for new data
  if (myImager.isDataReady() == true)
  {
    if (myImager.getRangingData(&measurementData)) //Read distance data into array
    {
      //The ST library returns the data transposed from zone mapping shown in datasheet
      //Pretty-print data with increasing y, decreasing x to reflect reality
      for (int y = 0 ; y <= imageWidth * (imageWidth - 1) ; y += imageWidth)
      {
        for (int x = imageWidth - 1 ; x >= 0 ; x--)
        {
          Serial.print("\t");
          Serial.print(measurementData.distance_mm[x + y]);
        }
        Serial.println();
      }
      Serial.println();
    }
  }

  delay(5); //Small delay between polling
}
```

### Error/Debug Message

```plain
Version with added delay(which should work):

== Rtl8735b IoT Platform ==

[Normal mode]
BootFromNORFlash
[Start Boot ROM...]
=== Load PARTBL ===
=== Load Done ===
=== Load ISP_IQ ===
[fcs chk pass]
ISP_IQ @ 0x8461080, 0x14f80, 0x0
mfcs_data version 0x00010001
fcs_data version 0x00010101
=== Process ISP_IQ ===
=== Load Done ===
=== Load BL ===
[Image Start Table @ 0x18200]
=== Load Done ===

== Boot Loader ==
Oct 24 2024:17:41:10
=== Load FCS Para ===
=== Load Done ===
[crc pass]
=== Load ISP_IQ Sensor ===
ISP_IQ @ 0x8461080, 0x14f80
=== Process ISP_IQ ===
=== Load Done ===
=== Load FW1 ===
[Image Start Table @ 0x20106200]
RAM Load @ 0x8061100->0x20106200, 0x5d4c
DRAM_TYPE is DDR2 128MB.
ddr_freq = 533
DDR Load @ 0x8067080->0x70100000, 0x2cf64
=== FW Load Done ===

Boot Loader <==

== RAM Start ==
Build @ 17:32:16, Oct 24 2024

$8735b>SparkFun VL53L5CX Imager Example
Initializing sensor board. This can take up to 10s. Please wait.
```

### I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

- [x] I confirm I have checked online documentation, FAQ, GitHub Wiki and existing/closed issues.

M-ichae-l, 2025-06-12:
We're actively working on integrating ST VL53L5CX/VL53L7CX support in the upcoming release—stay tuned!

pammyleong, 2025-09-01:
Hi @electriczity ,

You can find VL53L5CX examples here: https://github.com/Ameba-AIoT/ameba-arduino-pro2/tree/dev/Arduino_package/hardware/libraries/Wire/examples/VL53L5CX

closed 2025-09-01 by pammyleong