Ameba-AIoT/ameba-arduino-pro2 #303: H265 Encoding in RTSP Streaming

deepakbinwal97, 2025-02-20:
### Hardware Environment

AMB82-Mini

### Related technical area

WiFi

### Is the feature request related to an un-fixed issue?

NO

### Describe the ideal solution/reault/output.

Hi , I want to know how to enable H265 encoding in arduino RTSP audio + video streaming and also how to use rtmp / webrtc to stream video+audio remotely

### I have checked existing/closed list of issues and the Contributing Guide.

- [x] I confirm I have checked existing/closed list of issues and the Contributing Guide.

ouoam, 2025-04-21:
I can use HEVC in 4.0.7. But 4.0.8 can not

kevinlookl, 2025-06-12:
Hi @deepakbinwal97 ,

You may use `VideoSetting(uint8_t resolution, uint8_t fps, uint8_t encoder, uint8_t snapshot);` for customized encoder type. For example, you may predefine it in .ino using `VideoSetting configV(VIDEO_FHD, 30, VIDEO_HEVC, 0);`

For RTMP streaming, you may refer to this [post](https://forum.amebaiot.com/t/sharing-video-streaming-from-ameba-pro2-to-browser-via-ffmpeg/1505/3) on forum. 

Thank you.