Ameba-AIoT/ameba-arduino-pro2 #320: The "Wire.h" library by Amb82 differs from the "Wire.h" library by AVR

Hienvn2222, 2025-04-27:
### Hardware Environment

AMB82 - i2c...

### Related technical area

I2C...

### Is the feature request related to an un-fixed issue?

The incompatibility error with sensor circuits using I2C requires redeclaring the original library from scratch.

### Describe the ideal solution/reault/output.

There is no need to redeclare the original library for each type of sensor board using I2C connection.

### I have checked existing/closed list of issues and the Contributing Guide.

- [x] I confirm I have checked existing/closed list of issues and the Contributing Guide.

geofrancis, 2026-01-08:
I can use the same i2c device libraries with AVR, ESP32 and STM32 its only AMB82-mini that is incompatible and needs special arduino libraries for each device.

M-ichae-l, 2026-01-09:
All these boards (AVR, ESP32, STM32, Ameba) use the Arduino Wire API for I²C.

But each platform actually has its own implementation of Wire.h that talks to the hardware differently.

AVR’s Wire talks to AVR TWI hardware

ESP32’s Wire talks to ESP32’s I²C

STM32’s Wire talks to STM32’s I²C

Ameba’s Wire talks to Ameba’s I²C driver

Even though the API functions look the same, the **behavior and low-level details can differ enough that some sensor libraries that assume AVR-style I²C behavior fail on Ameba.

❌ “Only AMB82-mini is incompatible and needs special libraries”
✔ “Most standard Arduino I²C device libraries can work on AVR, ESP32, STM32, and Ameba if they use the common Wire API, but some libraries assume AVR behavior and may fail on Ameba’s different implementation, which requires adaptation or special versions.”

Ameba boards cannot use AVR libraries because AVR libraries are written specifically for 8-bit AVR microcontrollers and directly depend on AVR CPU instructions, registers, memory maps, and peripherals, while Ameba boards use completely different 32-bit ARM-based architectures with different hardware designs, so even though both can be programmed using the Arduino IDE, the underlying hardware is incompatible and AVR-specific code cannot compile or run on Ameba boards.